bulletphysics/bullet3 #2867: Collision Shape of the Object

TFLQW, 2020-06-18:
I have loaded an urdf file of an object in the environment, I try to test the collision shape of an peg by using a small sphere in the pybullet_data. The collision tag in urdf is .stl file.
The test result is like the figure: I find that it still have a little distance when they contact, could you tell me how solve this problem?
 
[image]

lucasrm25, 2020-06-22:
I think pyBulet uses a convex hull for mesh collision detection.
Take a look at the **loadURDF** method in the documentation.

If you want to detect concave meshes I think you just have to change your URDF like this example [samurai.urdf](https://github.com/bulletphysics/bullet3/blob/master/data/samurai.urdf) (be aware that you have to change the flag concave="yes" in two different places).

But this will slow down your simulation

erwincoumans, 2020-06-22:
For moving concave objects, it is best to use v-hacd convex decomposition. See 
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/vhacd.py
how to create a convex decomposition obj from a concave obj.
Then replace the obj in the urdf file.

For static concave triangle meshes, indeed you can use the concave="yes" flag, it should perform well, but doesn't support moving concave meshes.

closed 2020-06-22 by erwincoumans

robin-gdwl, 2023-05-30:
Nevermind, I figured out what my issue was: 
The VHACD decomposition that I had created did not have the faces as vertex groups as you explained[ here](https://github.com/bulletphysics/bullet3/issues/2692#issuecomment-604411825).
I used a different tool to create it than the oneintegrated into pybullet. 
After creating the vhacd decomposition again with the [script you provided ](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/vhacd.py). everything worked as expected. 

Thank you very much for your work.