bulletphysics/bullet3 #2869: Problem with torque control on prismatic joints.

Knightowl3128, 2020-06-18:
I have made a simple urdf model from solidworks consisting of a spherical mass supported by two prismatic legs. I have attached the URDF file with the corresponding meshes. I have also modified the urdf file to restrict the motion of the robot in only z-axis.

Now, I want to control the prismatic legs through torque control. So, to get the actual force values to support on one leg, I used position control for full extension on one leg and found the force required for support through getJointState. After that, I used this force value for torque control. This is the code I used.

```
import pybullet as p
import pybullet_data

physicsClient = p.connect(p.GUI)

p.setAdditionalSearchPath(pybullet_data.getDataPath())  # optionally
p.setGravity(0, 0, -10)
planeId = p.loadURDF("plane.urdf")
cubeStartPos = [0, 0,-1.2]

cubeStartOrientation = p.getQuaternionFromEuler([0, 0, 0])

boxId = p.loadURDF(r"Your Path\test_urdf2\urdf\slip_1d.urdf",
                   cubeStartPos, cubeStartOrientation)

p.setRealTimeSimulation(1)

# For support on only one leg, the other leg is moved to the horizontal position
p.setJointMotorControl2(boxId,5,controlMode=p.POSITION_CONTROL,targetPosition =  1.5,force = 5000)
while True:

    p.setJointMotorControl2(boxId, 3, p.VELOCITY_CONTROL, force=0)

    position = 0
    
    #Uncomment the below region to get torque values for support on one leg
    # p.setJointMotorControl2(boxId,3,controlMode=p.POSITION_CONTROL,targetPosition = position,force = 5000)
    #print('left',p.getJointState(boxId, 3)[3])

#torque value with force found from uncommenting the above region
    p.setJointMotorControl2(boxId, 3, controlMode=p.TORQUE_CONTROL, force=-881.8348310663498
                            )
```
However, with this force value the leg is not fully extended and does not support the body. I am not getting what I'm doing wrong here.

Also, I don't understand why the force value is above 800N when the spherical body mass is only 62 Kg and thus the force should be close to 62*10 = 620N.

I feel I'm missing something very obvious and any help would be appreciated.

erwincoumans, 2020-06-18:
You should not use 'realTimeSimulation' when using force control:
interleave the application of force with one stepSimulation call. See attached.
[test.zip](https://github.com/bulletphysics/bullet3/files/4801520/test.zip)

closed 2020-06-18 by erwincoumans

Knightowl3128, 2020-06-19:
Thank you so much for your quick reply. It would be nice if you can update the docs about this. I don't think it is mentioned there.