bulletphysics/bullet3 #2871: Differences in DeepMimic between bullet and pybullet versions

tfederico, 2020-06-19:
Hi,

which are the current differences between the DeepMimic implementation in pybullet and the original one?

erwincoumans, 2020-06-19:
They should be very similar, I tried using the same training code from Jason Peng, some differences include:
1) PyBullet uses a larger time step
2) PyBullet uses a URDF definition of the character, but it should be identical
3) PyBullet is easier to install (pip3 install pybullet)

There could be small other unintended differences. We could use the original policies from DeepMimic directly in PyBullet, but a while back some of them started to perform poorly (didn't have time to track down why) so I retrained some of the policies.

Closing for now, just reply if you have other questions.
Also, note that Jason Peng did an internship with us and we released another imitation environment based on PyBullet for our quadruped robots. See
https://github.com/google-research/motion_imitation

closed 2020-06-19 by erwincoumans

tfederico, 2020-06-22:
Looks amazing, thanks! I am currently interested in human-like characters, but I will definitively take a look at the paper :)

tfederico, 2020-06-23:
out of curiosity, is there any reason why you chose VideoPose3D instead of using HMR or HMMR like in the Skills from Video paper?

erwincoumans, 2020-06-24:
@tfederico There is no video involved in PyBullet deep_mimic.

tfederico, 2020-06-25:
sorry, I must have confused with some other repo I was looking at