bulletphysics/bullet3 #2880: Problem about slow rendering time

baoyufuyou, 2020-06-23:
Hi guys, I faced also problem of low rendering time. I tested the speed in my computer. Then, the getscreenpixels20423 will take about 17 ms.
with GPU 0: GeForce RTX 2070. Is that normal????? I'm a little bit confused.
Thanks a lot. Guys.

[image]

[image]
[image]

closed 2020-06-23 by baoyufuyou

erwincoumans, 2020-06-23:
It depends on resolution etc. Did you try to use EGL? Also, make sure to disable v-sync (rendering may stall on it).

It was closed, did you find a solution?

baoyufuyou, 2020-06-24:
Thank you very much.  ;) I found I used models with high quality and wrong projection matrix. I simlified the model and fix the projection matrix.   Then, the rendering looks fine. 
I will try to use EGL and test the v-sync. See will it be faster.