bulletphysics/bullet3 #2890: Linker issue with CMake

aravindsiv, 2020-06-29:
Hi,

I am the author of #2758 but I am unable to re-open the issue. I tried the fix that @erwincoumans  suggested, and I am now facing some linker issues. My CMake/Make commands are:

```
cmake -E env CXXFLAGS="-fPIC" cmake -DBUILD_PYBULLET=OFF -DBUILD_PYBULLET_NUMPY=OFF DUSE_DOUBLE_PRECISION=ON -DBT_USE_EGL=ON -DCMAKE_INSTALL_PREFIX=/path/to/app/ .. || exit 1
make install -j $(command nproc 2>/dev/null || echo 12) || exit 1
```

When I am linking the libraries in the suggested order inside my application, I get the following error message:

```
/path/to/app/lib/libBulletRobotics.a(b3PluginManager.o): In function `b3PluginManager::unloadPlugin(int)':
b3PluginManager.cpp:(.text+0xdc7): undefined reference to `dlclose'
/path/to/app/lib/libBulletRobotics.a(b3PluginManager.o): In function `b3PluginManager::registerStaticLinkedPlugin(char const*, b3PluginFunctions&, bool)':
b3PluginManager.cpp:(.text+0x164c): undefined reference to `dlclose'
/path/to/app/lib/libBulletRobotics.a(b3PluginManager.o): In function `b3PluginManager::loadPlugin(char const*, char const*)':
b3PluginManager.cpp:(.text+0x2e12): undefined reference to `dlopen'
b3PluginManager.cpp:(.text+0x356a): undefined reference to `dlsym'
b3PluginManager.cpp:(.text+0x357e): undefined reference to `dlsym'
b3PluginManager.cpp:(.text+0x3592): undefined reference to `dlsym'
b3PluginManager.cpp:(.text+0x35a6): undefined reference to `dlsym'
b3PluginManager.cpp:(.text+0x35ba): undefined reference to `dlsym'
/path/to/app/lib/libBulletRobotics.a(b3PluginManager.o):b3PluginManager.cpp:(.text+0x35ce): more undefined references to `dlsym' follow
/path/to/app/lib/libBulletRobotics.a(b3PluginManager.o): In function `b3PluginManager::loadPlugin(char const*, char const*)':
b3PluginManager.cpp:(.text+0x36f2): undefined reference to `dlclose'
b3PluginManager.cpp:(.text+0x3a30): undefined reference to `dlclose'
b3PluginManager.cpp:(.text+0x3ab6): undefined reference to `dlerror'
b3PluginManager.cpp:(.text+0x3e42): undefined reference to `dlclose'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `threadFunction(void*)':
b3PosixThreadSupport.cpp:(.text+0x2c): undefined reference to `sem_post'
b3PosixThreadSupport.cpp:(.text+0x42): undefined reference to `sem_wait'
b3PosixThreadSupport.cpp:(.text+0x56): undefined reference to `sem_wait'
b3PosixThreadSupport.cpp:(.text+0x94): undefined reference to `sem_post'
b3PosixThreadSupport.cpp:(.text+0xa8): undefined reference to `sem_post'
b3PosixThreadSupport.cpp:(.text+0x10c): undefined reference to `sem_post'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixThreadSupport::waitForResponse(int*, int*)':
b3PosixThreadSupport.cpp:(.text+0x158): undefined reference to `sem_wait'
b3PosixThreadSupport.cpp:(.text+0x1d6): undefined reference to `sem_wait'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixThreadSupport::isTaskCompleted(int*, int*, int)':
b3PosixThreadSupport.cpp:(.text+0x252): undefined reference to `sem_trywait'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixThreadSupport::runTask(int, void*, int)':
b3PosixThreadSupport.cpp:(.text+0x30c): undefined reference to `sem_post'
b3PosixThreadSupport.cpp:(.text+0x32c): undefined reference to `sem_post'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixThreadSupport::createBarrier()':
b3PosixThreadSupport.cpp:(.text+0x3a8): undefined reference to `pthread_barrier_init'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixThreadSupport::deleteBarrier(b3Barrier*)':
b3PosixThreadSupport.cpp:(.text+0x44c): undefined reference to `pthread_barrier_destroy'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixThreadSupport::stopThreads()':
b3PosixThreadSupport.cpp:(.text+0x495): undefined reference to `sem_wait'
b3PosixThreadSupport.cpp:(.text+0x4b5): undefined reference to `sem_destroy'
b3PosixThreadSupport.cpp:(.text+0x4c9): undefined reference to `sem_destroy'
b3PosixThreadSupport.cpp:(.text+0x50d): undefined reference to `pthread_join'
b3PosixThreadSupport.cpp:(.text+0x555): undefined reference to `sem_post'
b3PosixThreadSupport.cpp:(.text+0x56e): undefined reference to `sem_post'
b3PosixThreadSupport.cpp:(.text+0x59c): undefined reference to `sem_wait'
b3PosixThreadSupport.cpp:(.text+0x5b5): undefined reference to `sem_wait'
b3PosixThreadSupport.cpp:(.text+0x5f7): undefined reference to `pthread_join'
b3PosixThreadSupport.cpp:(.text+0x63c): undefined reference to `sem_destroy'
b3PosixThreadSupport.cpp:(.text+0x650): undefined reference to `sem_destroy'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixThreadSupport::startThreads(b3PosixThreadSupport::ThreadConstructionInfo&)':
b3PosixThreadSupport.cpp:(.text+0x72e): undefined reference to `sem_init'
b3PosixThreadSupport.cpp:(.text+0x7f4): undefined reference to `sem_init'
b3PosixThreadSupport.cpp:(.text+0x811): undefined reference to `sem_init'
b3PosixThreadSupport.cpp:(.text+0x854): undefined reference to `pthread_create'
b3PosixThreadSupport.cpp:(.text+0x87d): undefined reference to `pthread_create'
b3PosixThreadSupport.cpp:(.text+0x8d0): undefined reference to `sem_init'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixBarrier::setMaxCount(int)':
b3PosixThreadSupport.cpp:(.text._ZN14b3PosixBarrier11setMaxCountEi[_ZN14b3PosixBarrier11setMaxCountEi]+0x14): undefined reference to `pthread_barrier_init'
/path/to/app/lib/libBulletRobotics.a(b3PosixThreadSupport.o): In function `b3PosixBarrier::sync()':
[... 52 lines cut ...]
egl.c:(.text+0xeda): undefined reference to `dlsym'
/path/to/app/lib/libOpenGLWindow.a(egl.o):egl.c:(.text+0xef7): more undefined references to `dlsym' follow
/path/to/app/lib/libOpenGLWindow.a(egl.o): In function `gladLoaderLoadEGL':
egl.c:(.text+0xf51): undefined reference to `dlclose'
egl.c:(.text+0x100f): undefined reference to `dlsym'
egl.c:(.text+0x102c): undefined reference to `dlsym'
egl.c:(.text+0x1049): undefined reference to `dlsym'
egl.c:(.text+0x1066): undefined reference to `dlsym'
egl.c:(.text+0x1083): undefined reference to `dlsym'
egl.c:(.text+0x10ad): undefined reference to `dlopen'
egl.c:(.text+0x10c7): undefined reference to `dlopen'
egl.c:(.text+0x1129): undefined reference to `dlsym'
egl.c:(.text+0x1146): undefined reference to `dlsym'
egl.c:(.text+0x1163): undefined reference to `dlsym'
egl.c:(.text+0x1180): undefined reference to `dlsym'
egl.c:(.text+0x119d): undefined reference to `dlsym'
/path/to/app/lib/libOpenGLWindow.a(egl.o):egl.c:(.text+0x11ba): more undefined references to `dlsym' follow
/path/to/app/lib/libOpenGLWindow.a(egl.o): In function `gladLoaderUnloadEGL':
egl.c:(.text+0x1a81): undefined reference to `dlclose'
collect2: error: ld returned 1 exit status
```

I tried the fix suggested in #1236, but that doesn't work for me. Any help would be appreciated.

 Thanks in advance!

erwincoumans, 2020-06-29:
Did you add the 'dl' to the link libraries? (either using -ldl or using CMAKE_DL_LIBS)

aravindsiv, 2020-06-29:
Thanks, @erwincoumans  - that worked!

closed 2020-06-29 by aravindsiv