bulletphysics/bullet3 #2891: No setJointMotorControlMultiDof for right shoulder in setJointMotors

tfederico, 2020-06-29:
Hi,

I noticed that in [humanoid_stable_pd.py](https://github.com/bulletphysics/bullet3/blob/0aaae872451a69d0c93b0c8ed818667de4ad5653/examples/pybullet/gym/pybullet_envs/deep_mimic/env/humanoid_stable_pd.py) there is a call to `pybullet.setJointMotorControlMultiDof`for basically every joint, except for the right shoulder.

Is this on purpose? Why is it the only one left out?

erwincoumans, 2020-06-29:
It is not on purpose. The code is not used, right? Since by default we use stable PD control (setJointMotors is not called).

tfederico, 2020-06-30:
I see, thanks for the clarification. Is a refactoring of code happening in future releases? Or would a pull request with a refactoring be accepted? I see a lot of development code that could potentially be removed, making the whole flow more readable.

closed 2021-04-26 by erwincoumans