bulletphysics/bullet3 #2893: Half Cheetah Falls Through Heightfield

oisinturbitt, 2020-06-30:
Hi There,

I am attempting to use the GEOM_HEIGHTFIELD collision shape to generate a random heightfield using Perlin noise. This heightfield is then used to train a half cheetah walker. However, it occurs occasionally that when the half cheetah spawns, a limb will either penetrate the heightfield mesh causing the half cheetah to get stuck or fall through the heightfield entirely.

I have tried using different mesh sizings along changing the numSolverIterations and contactERP parameters but this behaviour still occurs. There is some images of this happening below.

The build I am using is on Ubuntu 18.04, using Python 3.7.4 and pyBullet 2.8.0.

[image]
[image]

erwincoumans, 2020-06-30:
Spawn the half-cheeta above the heightfield, not inside. You can perform a collision query and move the robot down (spawn at a safe height).

oisinturbitt, 2020-06-30:
Hi Erwin, 

Thanks for taking the time to get back. We are spawning the half cheetah well above the heightfield and it is falling down on to it. Still, the half-cheetah will either get stuck in the heightfield or fall right through it.

erwincoumans, 2020-06-30:
Can you provide a small reproduction case, preferably no other dependencies?

oisinturbitt, 2020-07-06:
I was able to find the bug and it was as you said. Due to the random terrain generation, if we had generated terrain with a large incline that the half-cheetah was reset far too high which then lead to the cheetah falling through. Seems to be fixed now and is working perfectly. Thanks a million for the help and for putting the library together!

closed 2020-07-06 by oisinturbitt