bulletphysics/bullet3 #2897: Building pybullet off of trunk leads to core dump after pybullet import

mt-, 2020-07-01:
Ubuntu 16.04

I can build pybullet without issue using the command:  ./build_cmake_pybullet_double.sh

However when I set my PYTHONPATH env variable, start a python 3.5.2 session, and import pybullet, I get a core dump when I exit the session.  pybullet also core dumps when I run it standalone as part of a python script.

I need to be able to build from source so I can increase MAX_DEGREE_OF_FREEDOM to 256 (or else my urdf won't load).  Is there an up to date stable branch I can build from?

Core dump below

*** Error in `python3': double free or corruption (fasttop): 0x0000000004f67ca0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fcb3af0a7e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7fcb3af1337a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7fcb3af1753c]
/lib/x86_64-linux-gnu/libc.so.6(+0x39ff8)[0x7fcb3aeccff8]
/lib/x86_64-linux-gnu/libc.so.6(+0x3a045)[0x7fcb3aecd045]
python3[0x623edf]
python3[0x623fca]
python3(PyErr_PrintEx+0x36)[0x624036]
python3(PyRun_SimpleFileExFlags+0x1d9)[0x6252f9]
python3(Py_Main+0x456)[0x63e616]
python3(main+0xe1)[0x4d1761]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7fcb3aeb3830]
python3(_start+0x29)[0x5d57c9]
======= Memory map: ========
00400000-007a8000 r-xp 00000000 08:02 10486813                           /usr/bin/python3.5
009a7000-009a9000 r--p 003a7000 08:02 10486813                           /usr/bin/python3.5
009a9000-00a40000 rw-p 003a9000 08:02 10486813                           /usr/bin/python3.5
00a40000-00a71000 rw-p 00000000 00:00 0 
02741000-10533000 rw-p 00000000 00:00 0                                  [heap]
7fcaa5934000-7fcaa5d34000 rw-s 00000000 00:06 538                        /dev/nvidiactl
7fcacbc00000-7fcacc000000 rw-s 00000000 00:06 538                        /dev/nvidiactl
7fcacc000000-7fcacc9f1000 rw-p 00000000 00:00 0 
7fcacc9f1000-7fcad0000000 ---p 00000000 00:00 0 
7fcad0336000-7fcad0536000 rw-s 00000000 00:06 538                        /dev/nvidiactl
7fcad2ec3000-7fcad32c3000 rw-s 00000000 00:06 538                        /dev/nvidiactl
7fcad42e1000-7fcad42e2000 ---p 00000000 00:00 0 
7fcad42e2000-7fcad4ae2000 rw-p 00000000 00:00 0 
7fcad4d03000-7fcad5103000 rw-s 00000000 00:06 538                        /dev/nvidiactl
7fcad5930000-7fcad5b30000 rw-s 00000000 00:06 538                        /dev/nvidiactl
7fcade331000-7fcadf409000 r-xp 00000000 08:02 22161716                   /usr/lib/nvidia-396/libnvidia-glcore.so.396.45
7fcadf409000-7fcadf42a000 rwxp 010d8000 08:02 22161716                   /usr/lib/nvidia-396/libnvidia-glcore.so.396.45
7fcadf42a000-7fcadfae5000 r-xp 010f9000 08:02 22161716                   /usr/lib/nvidia-396/libnvidia-glcore.so.396.45
7fcadfae5000-7fcadfce4000 ---p 017b4000 08:02 22161716                   /usr/lib/nvidia-396/libnvidia-glcore.so.396.45
7fcadfce4000-7fcadffe6000 rw-p 017b3000 08:02 22161716                   /usr/lib/nvidia-396/libnvidia-glcore.so.396.45
7fcadffe6000-7fcae0000000 rw-p 00000000 00:00 0 
7fcae0000000-7fcae2897000 rw-p 00000000 00:00 0 
7fcae2897000-7fcae4000000 ---p 00000000 00:00 0 
7fcae4074000-7fcae40f4000 rw-s 00000000 00:06 538                        /dev/nvidiactl
7fcae41f4000-7fcae4204000 r-xp 00000000 08:02 10492921                   /usr/lib/x86_64-linux-gnu/libdrm.so.2.4.0
7fcae4204000-7fcae4403000 ---p 00010000 08:02 10492921                   /usr/lib/x86_64-linux-gnu/libdrm.so.2.4.0
7fcae4403000-7fcae4404000 r--p 0000f000 08:02 10492921                   /usr/lib/x86_64-linux-gnu/libdrm.so.2.4.0
7fcae4404000-7fcae4405000 rw-p 00010000 08:02 10492921                   /usr/lib/x86_64-linux-gnu/libdrm.so.2.4.0
7fcae4405000-7fcae450a000 rw-s 00000000 00:06 538                        /dev/nvidiactl
7fcae450a000-7fcae451b000 r-xp 00000000 08:02 10493687                   /usr/lib/x86_64-linux-gnu/libXext.so.6.4.0
7fcae451b000-7fcae471a000 ---p 00011000 08:02 10493687                   /usr/lib/x86_64-linux-gnu/libXext.so.6.4.0
7fcae471a000-7fcae471b000 r--p 00010000 08:02 10493687                   /usr/lib/x86_64-linux-gnu/libXext.so.6.4.0
7fcae471b000-7fcae471c000 rw-p 00011000 08:02 10493687                   /usr/lib/x86_64-linux-gnu/libXext.so.6.4.0
7fcae471c000-7fcae471f000 r-xp 00000000 08:02 22287585                   /usr/lib/nvidia-396/tls/libnvidia-tls.so.396.45
7fcae471f000-7fcae491f000 ---p 00003000 08:02 22287585                   /usr/lib/nvidia-396/tls/libnvidia-tls.so.396.45
7fcae491f000-7fcae4920000 rw-p 00003000 08:02 22287585                   /usr/lib/nvidia-396/tls/libnvidia-tls.so.396.45
7fcae4920000-7fcae49e9000 r-xp 00000000 08:02 22161699                   /usr/lib/nvidia-396/libGL.so.396.45
7fcae49e9000-7fcae4a1d000 rwxp 000c9000 08:02 22161699                   /usr/lib/nvidia-396/libGL.so.396.45
7fcae4a1d000-7fcae4a31000 r-xp 000fd000 08:02 22161699                   /usr/lib/nvidia-396/libGL.so.396.45
7fcae4a31000-7fcae4c31000 ---p 00111000 08:02 22161699                   /usr/lib/nvidia-396/libGL.so.396.45
7fcae4c31000-7fcae4c56000 rw-p 00111000 08:02 22161699                   /usr/lib/nvidia-396/libGL.so.396.45
7fcae4c56000-7fcae4c5d000 rw-p 00000000 00:00 0 
7fcae4c5d000-7fcae4c5e000 ---p 00000000 00:00 0 
7fcae4c5e000-7fcae545e000 rw-p 00000000 00:00 0 
[... 815 lines cut ...]
7fffb8f4a000-7fffb8f6c000 rw-p 00000000 00:00 0                          [stack]
7fffb8ff9000-7fffb8ffc000 r--p 00000000 00:00 0                          [vvar]
7fffb8ffc000-7fffb8ffe000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
[mt:27983] *** Process received signal ***
[mt:27983] Signal: Aborted (6)
[mt:27983] Signal code:  (-6)
[mt:27983] [ 0] /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390)[0x7fcb3b26e390]
[mt:27983] [ 1] /lib/x86_64-linux-gnu/libc.so.6(gsignal+0x38)[0x7fcb3aec8428]
[mt:27983] [ 2] /lib/x86_64-linux-gnu/libc.so.6(abort+0x16a)[0x7fcb3aeca02a]
[mt:27983] [ 3] /lib/x86_64-linux-gnu/libc.so.6(+0x777ea)[0x7fcb3af0a7ea]
[mt:27983] [ 4] /lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7fcb3af1337a]
[mt:27983] [ 5] /lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7fcb3af1753c]
[mt:27983] [ 6] /lib/x86_64-linux-gnu/libc.so.6(+0x39ff8)[0x7fcb3aeccff8]
[mt:27983] [ 7] /lib/x86_64-linux-gnu/libc.so.6(+0x3a045)[0x7fcb3aecd045]
[mt:27983] [ 8] python3[0x623edf]
[mt:27983] [ 9] python3[0x623fca]
[mt:27983] [10] python3(PyErr_PrintEx+0x36)[0x624036]
[mt:27983] [11] python3(PyRun_SimpleFileExFlags+0x1d9)[0x6252f9]
[mt:27983] [12] python3(Py_Main+0x456)[0x63e616]
[mt:27983] [13] python3(main+0xe1)[0x4d1761]
[mt:27983] [14] /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7fcb3aeb3830]
[mt:27983] [15] python3(_start+0x29)[0x5d57c9]
[mt:27983] *** End of error message ***
Aborted (core dumped)

erwincoumans, 2020-07-01:
If you remove pybullet fully and install using "pip3 install pybullet --upgrade --user", do you get the same crashes?

mt-, 2020-07-01:
Nope, that works fine.  Looks to be just when I bulid from source on master.  It's odd though as I've built from source previously and never run into this issue.

mt-, 2020-07-01:
And, cloning --branch 2.89 works fine after building from source.  So the issue is definitely something to do with master.

mt-, 2020-07-01:
Anyways thanks for the help, I don't really need to be on master so I survive for now.  Just thought I'd report the issue, seems like it's reproducible for me so if similar reports pop up I'm happy to help diagnose.

erwincoumans, 2020-07-01:
It would be good to know why cmake builds fail, while pip works ok. Tracking down the commit causing it is time consuming though (git bisect?)

mt-, 2020-07-01:
Agreed, I'm out the rest of the week but I'll try and spend a couple hours tracking it down sometime in the next couple of weeks.  Shouldn't be that hard as the compile / test loop is pretty short, as you say it's really just tedious.  Still would be nice to know that pybullet isn't going to break in a future update (although this could finally be the catalyst that gets me to upgrade Ubuntu).

closed 2021-04-26 by erwincoumans