bulletphysics/bullet3 #2898: resetSimulation doesn't free memory

fxia22, 2020-07-01:
Hi Developers, 

When I try to reset the simulation and remove all objects, it seems that memory was not freed. Here is a code snippet to reproduce the bug:

```python
import logging
import os
import pybullet as p
import psutil
proc = psutil.Process(os.getpid())

for i in range(100):
    cid = p.connect(p.DIRECT)
    p.setPhysicsEngineParameter(enableFileCaching=0, physicsClientId=cid)

    collision_id = p.createCollisionShape(p.GEOM_MESH,
                                              fileName='textured_simple_vhacd.obj',
                                              meshScale=1,
                                              physicsClientId=cid)
    visual_id = p.createVisualShape(p.GEOM_MESH,
                                    fileName='textured_simple.obj',
                                    meshScale=1,
                                    physicsClientId=cid)

    body_id = p.createMultiBody(baseCollisionShapeIndex=collision_id,
                                baseVisualShapeIndex=visual_id,
                                physicsClientId=cid
                                )

    p.resetSimulation(physicsClientId=cid)
    p.disconnect(physicsClientId=cid)
    logging.critical(
        f"At iter {i}, memory usage after reload is {proc.memory_info().rss / (1024**3) :.2f} GB")
```
output:

```
pybullet build time: Jun 30 2020 23:52:50
CRITICAL:root:At iter 0, memory usage after reload is 0.10 GB
CRITICAL:root:At iter 1, memory usage after reload is 0.13 GB
CRITICAL:root:At iter 2, memory usage after reload is 0.18 GB
CRITICAL:root:At iter 3, memory usage after reload is 0.22 GB
CRITICAL:root:At iter 4, memory usage after reload is 0.27 GB
CRITICAL:root:At iter 5, memory usage after reload is 0.32 GB
CRITICAL:root:At iter 6, memory usage after reload is 0.36 GB
CRITICAL:root:At iter 7, memory usage after reload is 0.41 GB
CRITICAL:root:At iter 8, memory usage after reload is 0.46 GB
CRITICAL:root:At iter 9, memory usage after reload is 0.50 GB
CRITICAL:root:At iter 10, memory usage after reload is 0.55 GB
CRITICAL:root:At iter 11, memory usage after reload is 0.60 GB
```

If I change to `enableFileCaching=1` or `baseVisualShapeIndex=-1`, the memory leak goes away. Do you have any insights on why this happens? I am wondering is this the right use case of `resetSimulation`? Maybe this issue is related to #2216? Thanks a lot!

fxia22, 2020-07-01:
[003_cracker_box.zip](https://github.com/bulletphysics/bullet3/files/4856045/003_cracker_box.zip) Here is a copy of the data to reproduce the bug, in case it is relevant.

System information, ubuntu 18.04, Python 3.6.5, pybullet 2.8.3

sippeyxp, 2020-08-13:
Changed your example to use pybullet_data files and it still reproduces.

I progressively moved lines out of loop until the only command remains is createMultiBody and resetSimulation. The severeness of leak remain the same.

If I move resetSimulation out of loop, the leak is 4x faster in terms of memory.

Hope the information is helpful for debugging.

erwincoumans, 2020-08-13:
Thanks for the report, it was a texture not released, only in case of programmatic generation, fixed here:
https://github.com/bulletphysics/bullet3/pull/2987
(using loadURDF would not leak, or using file caching as you mentioned)

closed 2020-08-13 by erwincoumans

fxia22, 2020-08-14:
Thanks a lot Erwin!