bulletphysics/bullet3 #2906: Different results between resetJointState and setJointMotorControl2

seann999, 2020-07-01:
I am controlling a mobile manipulator in Pybullet.
Setting the joint positions (calculated with IK) with resetJointState results in a desired pose, but setting them with setJointMotorControl2 and running the simulation for N (>100) steps yields in a pose that is off (and thus the end effector is not in the target pose). I initially thought that it might be because of some collision, but that doesn't seem to be the cause. What could be it?

seann999, 2020-07-02:
Upon further investigation, it seems like setJointMotorControl2 does not guarantee that the final position will be at the target position.
What can I do to deal with this?

seann999, 2020-07-02:
Orange lines are target values, blue ones are actual positions
[image]

erwincoumans, 2020-07-02:
small self-contained repro case and I'll re-open the issue :)

closed 2020-07-02 by erwincoumans

seann999, 2020-07-08:
I've created a gist to reproduce the behavior:
https://gist.github.com/seann999/9f43f493cdce56ed87e662682e3660c4
These are the joint states produced (the last 2 plots for gripper states can be ignored):
[image]

erwincoumans, 2020-07-08:
I ran the script but there are no graphs.
It usually takes a couple of simulation steps (not just 1) for the motors to reach a target pose, how many steps did you allow? In addition, the gains/forces need to be tuned well.

erwincoumans, 2020-07-08:
I had a quick look, the robot doesn't have a fixed base and it penetrating the plane. Please remove the plane and check again. Also, you may want to use a few simulation steps to allow the control to reach the target.

seann999, 2020-07-09:
I hadn't noticed the penetration; removing the plane worked.
Thank you!