bulletphysics/bullet3 #2907: IK support for spherical/MultiDof joints

ManifoldFR, 2020-07-02:
Hello,

I'm trying to retarget motion to a custom robot hand URDF I made that has 10 revolute and 5 spherical (3 DoF) joints. I have a list of Numpy arrays `targets` where each entry is a vec3 
```python
joint_poses = pb.calculateInverseKinematics2(
    objUid,
    end_effector_idx,
    targetPositions=targets,
    maxNumIterations=10,
    solver=pb.IK_DLS
)
```
I get a tuple of length 25, which I guess is the number of DoFs (25 = 3 x 5 + 2 x 5) obtained from `btMultiBody::getNumDofs()`. I think that number should be 30 = 4 x 5 + 2 x 5 (the value of `btMultiBody::getNumPosVar()`, presumably), because the 5 spherical joints usually have their positions described by quaternions.

I guess there is no official support for this right now, then? I started looking at the source code, but I'm not sure if the actual base IK code supports 3 DoF joints.

erwincoumans, 2020-07-02:
IK for multidof joints is currently not supported indeed. Closing for now, if someone wants to implement it we can re-open/create a PR.

closed 2020-07-02 by erwincoumans

ManifoldFR, 2020-07-02:
Thanks ! I'll take a look at it if I have time later on.