bulletphysics/bullet3 #2908: calculateInverseKinematics returning poses violating joint limits

seann999, 2020-07-02:
Though I am not entirely sure how to use the optional parameters in the function as asked [here](https://github.com/bulletphysics/bullet3/issues/2902), I am passing lower, upper, and ranges of the movable joints (based on the URDF) to calculateInverseKinematics. However, the returned joint positions are not always within the limits provided. Is there no guarantee that the IK does not violate limits or am I possibly using it incorrectly?

Steven89Liu, 2020-07-03:
hope this paper below can help you. Could you provide a simple test?(https://www.researchgate.net/publication/242367247_Liegeois_A_Automatic_supervisory_control_of_the_configuration_and_behavior_of_multibody_mechanisms_IEEE_Trans_Syst_Man_Cybern_712_868-871)

seann999, 2020-07-08:
Thanks for the response.
I've written a gist to reproduce the limit violation:
https://gist.github.com/seann999/55d0493d6d863c1b34d03f58c8b3f98b#file-ik_test-py
For instance, the 2nd trial violates 2 revolute joint limits

seann999, 2020-07-09:
I think I found the problem. I was giving the IK function bad values; the passed lists (ll, ul, rp, jr) had different lengths.
In addition, the dof of the robot is 10 (as that is the length of the list returned by the function), so I should have passed it 4 lists of length 10, corresponding to each movable joint. Fixing this returned appropriate values which reach the given target position+orientation.

closed 2020-07-09 by seann999