bulletphysics/bullet3 #2909: calculateJacobian memory leak

Hayleymol, 2020-07-03:
I found a memory leak while running function p.calculateJacobian.
If I call calculateJacobian multiple times, the memory(RAM) increases limitlessly with fast speed. 
I guess it is natural for the code to use some RAM, but it increases sharply.
You can easily reproduce this problem by putting a while loop in the example code, jacobian.py (URL: https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/jacobian.py)

System information: 
Ubuntu 18.04, pybullet version: 2.8.3(it also happens in the previous version as well) , python version: 3.6.10

This is code when you put a while loop.
[image]

erwincoumans, 2020-08-25:
Thanks for the report, this should be resolved by now.

closed 2020-08-25 by erwincoumans

Hayleymol, 2020-10-25:
thanks a lot!