bulletphysics/bullet3 #2913: getMeshData does not return indices

TontonTremblay, 2020-07-07:
I am trying to render some basic shapes created in pybullet with an other renderer. The problem I came to is that the function `getMeshData` does not return the indices. It firsts return an integer representing how many vertices there are, then there is a list of vertices, but no indices. Anyone has an idea how to get around? 

Here is a snippet of code: 
```
import pybullet as p
import time
import pybullet_data

physicsClient = p.connect(p.DIRECT)#or p.DIRECT for non-graphical version
p.setAdditionalSearchPath(pybullet_data.getDataPath()) #optionally
p.setGravity(0,0,-10)
planeId = p.loadURDF("plane.urdf")
cubeStartPos = [0,0,1]
cubeStartOrientation = p.getQuaternionFromEuler([0,0,0])
boxId = p.loadURDF("r2d2.urdf",cubeStartPos, cubeStartOrientation)

for i in range(p.getNumJoints(boxId)):	
	name = p.getJointInfo(boxId,i)[1]
	nb, vertices = p.getMeshData(boxId,i)
	if nb < 1:
		continue
	print(name,nb,len(vertices))

```

DanielTakeshi, 2020-07-21:
@TontonTremblay  from my experience, it seems like vertices are always returned in a consistent order for the second item. Therefore you can just index it by the ordering in that second item.

seann999, 2021-01-19:
I still don't understand the data structure this is using... My mesh has 814 vertices, but how do I get the triplets?

DanielTakeshi, 2021-01-19:
@seann999  The API was recently updated. What code are you using?

seann999, 2021-01-19:
pybullet==3.0.8

```
count, vertices = p.getMeshData(body_id, link_idx)
vertices = np.array(vertices)
???
```

(My current goal is to convert this to an Open3D TriangleMesh)

seann999, 2021-01-19:
(Oops, deleted comment on accident)

pybullet==3.0.8

```
count, vertices = p.getMeshData(body_id, link_idx)
vertices = np.array(vertices)
???
```

(My goal is to convert this to an Open3D TriangleMesh)

Guptajakala, 2021-04-10:
@DanielTakeshi @TontonTremblay @seann999 I'm having the similar issue. Is there any solution for this?

DanielTakeshi, 2021-04-10:
@Guptajakala  what exactly are you trying to do? Can you paste a simple 10-line code snippet, then show why the result is not what you expected? Also please include the PyBullet version.

Guptajakala, 2021-04-10:
@DanielTakeshi 
Hi, I wanted to do the same thing as above, to get triangular indices as well. You mentioned that the vertex orders don't change in pybullet. Does it mean the only way is to maintain the triangular indices separately myself, by reading them from each .obj or .stl file in a urdf? That sounds a little bit inconvenient, when the urdf has multiple links.

My pybullet version is 3.0.8

```
import pybullet as p
import time
import pybullet_data

physicsClient = p.connect(p.DIRECT)#or p.DIRECT for non-graphical version
p.setAdditionalSearchPath(pybullet_data.getDataPath()) #optionally
p.setGravity(0,0,-10)
planeId = p.loadURDF("plane.urdf")
cubeStartPos = [0,0,1]
cubeStartOrientation = p.getQuaternionFromEuler([0,0,0])
boxId = p.loadURDF("r2d2.urdf",cubeStartPos, cubeStartOrientation)

for i in range(p.getNumJoints(boxId)):	
	name = p.getJointInfo(boxId,i)[1]
	nb, vertices = p.getMeshData(boxId,i)
	if nb < 1:
		continue
	print(name,nb,len(vertices))

```

erwincoumans, 2021-04-10:
The indices arw not exposed at the moment. Note that the rendering indices maybe different from physics indices (many collision shapes don't keep indices at all)

Guptajakala, 2021-04-10:
@erwincoumans 
Got it, is there any plan to expose at least the visual shape indices? Pybullet must be internally using them somewhere. Perhaps just additionally return the face indices for experimenta/advanced usage?

closed 2021-04-26 by erwincoumans