bulletphysics/bullet3 #2922: Chain of bodies that looks like a loop, is this possible?

DarioUrbina, 2020-07-09:
I want to join a child link (i.e. Link #10), to the base link (i.e. link #-1). This is generating discrepancies when trying to run the simulation.  

Since I cannot accomplish this with declared joints in my .urdf files, I have opted to restrict the motion of a body with a artificial joint (I am relying on collisions to do so). This is apparently a good solution. But when working with very complex bodies the collisions are not respected and the models behaves like in a chaotic way. 

Here you can see my code: Pendulum_exercise/Inverted_pendulum_tendon_actuation.py

erwincoumans, 2020-07-09:
You can create a fixed constraint (or point-to-point constraint etc) to create loops.

DarioUrbina, 2020-07-10:
Thank you for the reply, 

In the zip file you can find the files, the first to the Python code, the second to the URDF code: 

[Inverted_pendulum_tendon_actuation.zip](https://github.com/bulletphysics/bullet3/files/4900085/Inverted_pendulum_tendon_actuation.zip)

closed 2021-04-26 by erwincoumans