bulletphysics/bullet3 #2924: "Invalid GLX version" error in Windows Subsystem Linux

kpmoorse, 2020-07-10:
I am trying to get PyBullet up and running on a home computer running Windows Subsystem Linux, and upon running any of the included demos I receive the error message shown below. Other graphical applications run as expected in WSL and the demos run as expected in a Linux virtual machine--albeit at a low frame rate, as is expected with a VM.

```
$ python3 -m pybullet_envs.examples.enjoy_TF_HumanoidFlagrunHarderBulletEnv_v1_2017jul
pybullet build time: Jul  8 2020 18:24:53
current_dir=/home/<user>/.local/lib/python3.8/site-packages/pybullet_envs/bullet
/home/<user>/.local/lib/python3.8/site-packages/gym/logger.py:30: UserWarning: WARN: Box bound precision lowered by casting to float32
  warnings.warn(colorize('%s: %s'%('WARN', msg % args), 'yellow'))
startThreads creating 1 threads.
starting thread 0
started thread 0
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Invalid GLX version: major 1, minor 2
```

kpmoorse, 2020-07-14:
It turns out that the problem was not with PyBullet but with the X server that I was using to display on Windows Subsystem Linux.

I was able to get the PyBullet demos running by switching from Xming (OpenGL 1.2) to VcXsrv (OpenGL 1.4) and unchecking "Native opengl" in the "Extra settings" section of the X server startup wizard.

closed 2020-07-14 by kpmoorse

erwincoumans, 2020-07-14:
Interesting, can you copy and paste the full output if you run that command?
~~~~~~~~~~~
python -m pybullet_envs.examples.enjoy_TF_HumanoidFlagrunHarderBulletEnv_v1_2017jul
~~~~~~~~~~~
For comparison, here is the 'regular' Windows version:
~~~~~~~~~~~~~~~~
C:\WINDOWS\system32>python -m pybullet_envs.examples.enjoy_TF_HumanoidFlagrunHarderBulletEnv_v1_2017jul
pybullet build time: Jul 13 2020 10:16:17
current_dir=c:\python37\lib\site-packages\pybullet_envs\bullet
c:\python37\lib\site-packages\gym\logger.py:30: UserWarning: [33mWARN: Box bound precision lowered by casting to float32[0m
  warnings.warn(colorize('%s: %s'%('WARN', msg % args), 'yellow'))
starting thread 0
started testThreads thread 0 with threadHandle 00000000000002D4
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
Version = 4.6.0 NVIDIA 446.14
Vendor = NVIDIA Corporation
Renderer = GeForce RTX 2080/PCIe/SSE2
b3Printf: Selected demo: Physics Server
starting thread 0
~~~~~~~~~~~~~~~~

kpmoorse, 2020-07-14:
I installed WSL2 but ran into errors setting it as the default for new Linux distributions so I believe I am running WSL1.

On a second look, it seems that I was mixing up the OpenGL and GLX version strings in glxinfo; I am running GLX 1.4 and OpenGL 3.1.

Running the command you requested (only with python3 instead of python) gives the expected graphical output of a humanoid figure running toward a red ball, and the following terminal output:

pybullet build time: Jul  8 2020 18:24:53
current_dir=/home/<user>/.local/lib/python3.8/site-packages/pybullet_envs/bullet
/home/<user>/.local/lib/python3.8/site-packages/gym/logger.py:30: UserWarning: WARN: Box bound precision lowered by casting to float32
  warnings.warn(colorize('%s: %s'%('WARN', msg % args), 'yellow'))
startThreads creating 1 threads.
starting thread 0
started thread 0
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=VMware, Inc.
GL_RENDERER=llvmpipe (LLVM 10.0.0, 256 bits)
GL_VERSION=3.3 (Core Profile) Mesa 20.0.8
GL_SHADING_LANGUAGE_VERSION=3.30
pthread_getconcurrency()=0
Version = 3.3 (Core Profile) Mesa 20.0.8
Vendor = VMware, Inc.
Renderer = llvmpipe (LLVM 10.0.0, 256 bits)
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0
MotionThreadFunc thread started
ven = VMware, Inc.
ven = VMware, Inc.
/home/<user>/.local/lib/python3.8/site-packages/pybullet_envs/examples/enjoy_TF_HumanoidFlagrunHarderBulletEnv_v1_2017jul.py:58: DeprecationWarning: The truth value of an empty array is ambiguous. Returning False, but in future this will result in an error. Use `array.size > 0` to check that an array is not empty.
  if still_open == False:
X connection to :0 broken (explicit kill or server shutdown).

erwincoumans, 2020-07-14:
Thanks, I'm considering installing this myself. Is this using Ubuntu 18.04 or other distro?

kpmoorse, 2020-07-14:
Ubuntu 20.04 LTS

erwincoumans, 2020-07-15:
Thanks for the tips, got it all to work. Except the mouse pointer is invisible in the X11/XLaunch window.

KangriX, 2024-04-16:
Can I ask how to upgrade GLX from 1.2 to 2.0 and above? I'm about to run pybullet on the server to generate a graphical program to be transferred to the local computer via xming. The program runs with the same Invalid GLX Version problem, and I've searched for ways to upgrade it without success.
My configuration is as follows:
`server glx version string: 1.2 
client glx version string: 1.4
GLX version: 1.2
    Max core profile version: 4.5
    Max compat profile version: 3.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL version string: 3.1 Mesa 21.2.6
OpenGL shading language version string: 1.40.`