bulletphysics/bullet3 #2935: Possible memory leak in saveStateToMemory

aravindsiv, 2020-07-15:
Hi,

I am trying to use the C++ API to simulate the MIT racecar. My program was using up a lot of memory, and when I ran valgrind to debug, I found a large number of errors that looked like below:

```
==60092==    at 0x4C2DB8F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==60092==    by 0x871182: btAlignedAllocDefault(unsigned long, int) (in /path/to/my/executable)
==60092==    by 0x5C0534: btDefaultSerializer::finishSerialization() (in /path/to/my/executable)
==60092==    by 0x5B44D4: PhysicsServerCommandProcessor::processSaveStateCommand(SharedMemoryCommand const&, SharedMemoryStatus&, char*, int) (in /path/to/my/executable)
==60092==    by 0x5B5065: PhysicsServerCommandProcessor::processCommand(SharedMemoryCommand const&, SharedMemoryStatus&, char*, int) (in /path/to/my/executable)
==60092==    by 0x6850F0: PhysicsDirect::submitClientCommand(SharedMemoryCommand const&) (in /path/to/my/executable)
==60092==    by 0x5F2933: b3SubmitClientCommand (in /path/to/my/executable)
==60092==    by 0x5F29CF: b3SubmitClientCommandAndWaitStatus (in /path/to/my/executable)
==60092==    by 0x580E4F: b3RobotSimulatorClientAPI_NoDirect::saveStateToMemory() (in /path/to/my/executable)
```

This happens even if I am not doing anything overly complicated in my executable. I am just loading the URDF of the racecar and the plane, switching on gravity, stepping the simulation, and then making a call to `saveStateToMemory()`. The problem is when I perform many simulations and keep calling `saveStateToMemory()`, even if I remove the states from memory, the memory usage of the program still blows up.

aravindsiv, 2020-07-16:
I tried the following edits to `examples/SharedMemory/PhysicsServerCommandProcessor.cpp`  (similar to the fix merged in #2397 ):

Inside `PhysicsServerCommandProcessor::processSaveStateCommand( ... )`, I added the following lines before the return statement:

```
delete ser; 
delete bulletFile;
```

Unfortunately, the second line doesn't let me save states anymore, and only leaving the first line, i.e. deleting the `btDefaultSerializer` object still causes the memory leak described earlier.

erwincoumans, 2020-07-17:
There is no removeState exposed in C++. I added it here, please check it https://github.com/bulletphysics/bullet3/pull/2940

aravindsiv, 2020-07-18:
Hi @erwincoumans ,

Thank you for your response! I tried the fix, but the issue in `saveStateToMemory()` still persists. This is the output from valgrind:

```
==93906== Conditional jump or move depends on uninitialised value(s)
==93906==    at 0x6BCB95: bParse::bFile::findLibPointer(void*) (in /path/to/my/executable)
==93906==    by 0x6BD123: bParse::bFile::resolvePointersStructRecursive(char*, int, int, int) (in /path/to/my/executable)
==93906==    by 0x6BD39E: bParse::bFile::resolvePointersChunk(bParse::bChunkInd const&, int) (in /path/to/my/executable)
==93906==    by 0x6BD868: bParse::bFile::resolvePointers(int) (in /path/to/my/executable)
==93906==    by 0x6BDED5: bParse::bFile::parseInternal(int, char*, int) (in /path/to/my/executable)
==93906==    by 0x6BEE6F: bParse::btBulletFile::parse(int) (in /path/to/my/executable)
==93906==    by 0x5B4603: PhysicsServerCommandProcessor::processSaveStateCommand(SharedMemoryCommand const&, SharedMemoryStatus&, char*, int) (in /path/to/my/executable)
==93906==    by 0x5B5135: PhysicsServerCommandProcessor::processCommand(SharedMemoryCommand const&, SharedMemoryStatus&, char*, int) (in /path/to/my/executable)
==93906==    by 0x6851C0: PhysicsDirect::submitClientCommand(SharedMemoryCommand const&) (in /path/to/my/executable)
==93906==    by 0x5F2A03: b3SubmitClientCommand (in /path/to/my/executable)
==93906==    by 0x5F2A9F: b3SubmitClientCommandAndWaitStatus (in /path/to/my/executable)
==93906==    by 0x580F1F: b3RobotSimulatorClientAPI_NoDirect::saveStateToMemory() (in /path/to/my/executable)
```

What I think is happening: Even after I call `removeState()`, the memory al is not cleaned up until the end of `PhysicsServerCommandProcessor`'s lifetime. Is there a way to clean up the `savedState`'s `m_bulletFile` and `m_serializer` objects when we delete the state?

erwincoumans, 2020-07-18:
The memory is cleaned up. Provide a self containrd miminall cpp code (including int main(int argv,argc)), preferably based on
 https://github.com/bulletphysics/bullet3/blob/master/examples/RobotSimulator/RobotSimulatorMain.cpp

I added those lines in the main loop, and no leaks.

erwincoumans, 2020-07-18:
Attached is the program I tested, and didn't find any growing memory or leaks.
[RobotSimulatorMain.zip](https://github.com/bulletphysics/bullet3/files/4941770/RobotSimulatorMain.zip)

Please check if this program causes leaks. If not, make minimal modifications to reproduce the leak.
You can use premake to generate project files on WIndows, Mac OSX and Linux, for example
~~~~~~~~~~~~~~~~
cd build3
./premake4_linux64 gmake
cd gmake App_RobotSimulator
make -j
../../bin/App_RobotSimulator
~~~~~~~~~~~~~~~~

erwincoumans, 2020-07-22:
I'm happy to help if you can provide small reproduction case.

closed 2020-08-11 by erwincoumans