bulletphysics/bullet3 #2949: Raycasts sometimes go through convex shapes and hit the far side when BT_USE_DOUBLE_PRECISION is enabled.

jitspoe, 2020-07-22:
I'm not sure if this is a Godot-specific issue or a problem with Bullet, but I figured I'd post an issue here.  I'm using the Godot engine, and ran into some precision issues with kinematic bodies returning bad normals on collision, so I defined BT_USE_DOUBLE_PRECISION and rebuilt.  This significantly improved the kinematic collision issues but broke some raycasts.

I made a simple convex collision shape that was just a box and still had the issues with the raycast.

[image]
There's a raycast to detect if the player should be able to hang on walls, and I had to make it longer because it wasn't hitting anything.  When the length was increased, I noticed it was hitting on the FAR side of the object.  Green line is drawn from the collision point and out on the Z (back) axis.  Cyan line shows the normal of the raycast collision.  Red is the normal from the kinematic body collision.  What's interesting is that if I'm on the other side of the object, it hits the same side, so it's not ALWAYS going through.

Disabled double precision to verify that was what was causing the problem, and the raycast hit on the right side again:
[image]

Here is the relevant code from Godot, I think:
```
bool BulletPhysicsDirectSpaceState::intersect_ray(const Vector3 &p_from, const Vector3 &p_to, RayResult &r_result, const Set<RID> &p_exclude, uint32_t p_collision_mask, bool p_collide_with_bodies, bool p_collide_with_areas, bool p_pick_ray) {

	btVector3 btVec_from;
	btVector3 btVec_to;

	G_TO_B(p_from, btVec_from);
	G_TO_B(p_to, btVec_to);

	// setup query
	GodotClosestRayResultCallback btResult(btVec_from, btVec_to, &p_exclude, p_collide_with_bodies, p_collide_with_areas);
	btResult.m_collisionFilterGroup = 0;
	btResult.m_collisionFilterMask = p_collision_mask;
	btResult.m_pickRay = p_pick_ray;

	space->dynamicsWorld->rayTest(btVec_from, btVec_to, btResult);
	if (btResult.hasHit()) {
		B_TO_G(btResult.m_hitPointWorld, r_result.position);
		B_TO_G(btResult.m_hitNormalWorld.normalize(), r_result.normal);
		CollisionObjectBullet *gObj = static_cast<CollisionObjectBullet *>(btResult.m_collisionObject->getUserPointer());
		if (gObj) {
			r_result.shape = btResult.m_shapeId;
			r_result.rid = gObj->get_self();
			r_result.collider_id = gObj->get_instance_id();
			r_result.collider = 0 == r_result.collider_id ? NULL : ObjectDB::get_instance(r_result.collider_id);
		} else {
			WARN_PRINTS("The raycast performed has hit a collision object that is not part of Godot scene, please check it.");
		}
		return true;
	} else {
		return false;
	}
}
```

```
/// It performs an additional check allow exclusions.
struct GodotClosestRayResultCallback : public btCollisionWorld::ClosestRayResultCallback {
	const Set<RID> *m_exclude;
	bool m_pickRay;
	int m_shapeId;

	bool collide_with_bodies;
	bool collide_with_areas;

public:
	GodotClosestRayResultCallback(const btVector3 &rayFromWorld, const btVector3 &rayToWorld, const Set<RID> *p_exclude, bool p_collide_with_bodies, bool p_collide_with_areas) :
			btCollisionWorld::ClosestRayResultCallback(rayFromWorld, rayToWorld),
			m_exclude(p_exclude),
			m_pickRay(false),
			m_shapeId(0),
			collide_with_bodies(p_collide_with_bodies),
			collide_with_areas(p_collide_with_areas) {}

	virtual bool needsCollision(btBroadphaseProxy *proxy0) const;

	virtual btScalar addSingleResult(btCollisionWorld::LocalRayResult &rayResult, bool normalInWorldSpace) {
		if (rayResult.m_localShapeInfo)
			m_shapeId = rayResult.m_localShapeInfo->m_triangleIndex; // "m_triangleIndex" Is a odd name but contains the compound shape ID
		else
			m_shapeId = 0;
		return btCollisionWorld::ClosestRayResultCallback::addSingleResult(rayResult, normalInWorldSpace);
	}
};
```

erwincoumans, 2020-07-22:
Thanks for the report. I can't see the issue in the code. Can you please create a minimal but fully working example in Bullet (modify some example, load the geometry etc)? You can attach zip files to issues.

jitspoe, 2020-07-22:
I also posted an issue on the Godot github in case it's specific to that: https://github.com/godotengine/godot/issues/40586

I have a project example there: https://github.com/godotengine/godot/files/4957613/Test.Double.Precision.Raycast.zip
here's the Godot engine prebuilt with that define set:  https://github.com/godotengine/godot/files/4957620/godot.windows.opt.tools.32_double_precision_bullet.zip

The define should be set for everything.  I added `env_base.Append(CPPDEFINES=["BT_USE_DOUBLE_PRECISION"])` to the SCons file.

I'll give the gjk flag a test.  Is there a good starting sample I could use if the Godot thing is too complex?

jitspoe, 2020-07-22:
Quick test seems like the kF_UseGjkConvexCastRaytest flag fixes the issue.  Why is that not the default?

erwincoumans, 2020-07-22:
For PyBullet we use it as default. I'll have a look making it default.

jitspoe, 2020-07-23:
Is there any reason not to use the GJK convex cast?  Just curious if there were some performance benefits or something for the other one.

erwincoumans, 2020-07-23:
The subsimplex convex cast is a bit faster, but less accurate, it is best to use the gjk one.

Actually, for many shapes it is better and faster to use a specialized ray intersection implementation. We didn't do that for Bullet, convex shapes don't even have indices/triangles by default. You can implement a faster and more accurate custom ray-convex and ray-box.

closed 2021-04-26 by erwincoumans