bulletphysics/bullet3 #2955: Not seeing visual colours of UR5 robot

deepakraina99, 2020-07-25:
I am loading the UR5 robot using the following code. I am not able to see the visual colors of the UR5 robot. Can you tell me what is the reason for this? I am attaching my urdf and meshes file.

---Code
[urdf_meshes.zip](https://github.com/bulletphysics/bullet3/files/4976159/urdf_meshes.zip)

---
import pybullet as p
import pybullet_data

GRAVITY = -9.8
dt = 1e-3

# load pybullet physics engine
physicsClient = p.connect(p.GUI)
p.setAdditionalSearchPath(pybullet_data.getDataPath())
p.resetSimulation()
p.setGravity(0, 0, GRAVITY)
p.setTimeStep(dt)
p.setPhysicsEngineParameter(fixedTimeStep=dt, numSolverIterations=100, numSubSteps=10)

planeId = p.loadURDF("plane.urdf")

robotStartPos = [0.0, 0.0, 0.0]
robotStartOrientation = p.getQuaternionFromEuler([0.0, 0.0, 0.0])
urdfFilename = 'urdf/ur5.urdf'
robotId = p.loadURDF(urdfFilename, useFixedBase=True)
p.resetBasePositionAndOrientation(robotId,
                                  robotStartPos,
                                  robotStartOrientation)

Steven89Liu, 2020-07-28:
i dont know how the Gazebo get the color, but i think the color in pybullet is as expected as what you set in the urdf.

deepakraina99, 2020-07-28:
Yes, it does. But it does not import any colors from _.dae_ file. Whatever color we set in <material> tag in urdf, it changes accordingly. But if you see the ur5 robot in the gazebo, some of the sections of link are in blue color. If  I change the color for this link in urdf file, whole color will change. Please see the picture below.

[image]

#### Code
````
  <link name="shoulder_link">
    <visual>
      <geometry>
        <mesh filename="../meshes/ur5/visual/shoulder.obj"/>
      </geometry>
      <material name="LightGrey">
        <color rgba="0.7 0.7 0.7 1.0"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <mesh filename="../meshes/ur5/collision/shoulder.stl"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="3.7"/>
      <origin rpy="0 0 0" xyz="0.0 0.0 0.0"/>
      <inertia ixx="0.010267495893" ixy="0.0" ixz="0.0" iyy="0.010267495893" iyz="0.0" izz="0.00666"/>
    </inertial>
  </link>
````

Steven89Liu, 2020-07-28:
so do you mean that bullet doesn't use the color info in the .obj file?

deepakraina99, 2020-07-28:
Yes.

Steven89Liu, 2020-07-28:
are you sure the blue color in the gazebo comes from the .obj file? i don't find the any color info from the .obj file. do i miss something?

deepakraina99, 2020-07-28:
I think loading colors from .obj file does not work in bullet. Its limited to single color per link. See this post #1934

closed 2020-07-30 by deepakraina99