bulletphysics/bullet3 #2961: How to change the parameters of loadSoftBody at any timing

asaka-t, 2020-07-30:
I want to simulate artificial muscle in pybullet. 
After loading the artificial muscle as a soft body, I want to change the springElasticStiffness in order to move it as an actuator. 
Please tell me how to change the parameters of loadSoftBody at any timing.

I think changeDynamics is for rigid bodies and not for soft bodies. Is this correct?

erwincoumans, 2020-07-31:
I think it is currently not supported, but would be nice to have indeed.

asaka-t, 2020-08-14:
I added a function to the source code that modifies some parameters of loadSoftBody.
The attached video(https://1drv.ms/v/s!AhSJ3fHM8y8IiGmdTuNQHaJ850cw?e=znQbby) is an example of increasing springElasticStiffness.

I want to use this to simulate an artificial muscle, but currently I have to use createSoftBodyAnchor.

What should I do if I want to handle urdf data containing rigid body links and soft body links?

closed 2021-04-26 by erwincoumans