bulletphysics/bullet3 #2963: Issue with contact force in pybullet during sliding

deepakraina99, 2020-08-02:
I have implemented an impedance control in pybullet. That means, whenever some external force is applied on a robot, it behaves as a spring and damper system. Due to this force, it gets displaced from its position and returns to its target position when an external force is removed. The equation that is used is given below:

[image]

The video regarding the same is given below:
[image]

Now I am trying to implement the same control strategy in a scenario where robot is polishing a surface with some uncertainties (unevenness) on the surface. I moved the robot close to the table and try to move while it is in contact with table. I am getting the normal force during contact using the p.getContactPoints. But the robot suddenly takes a and leaves contact. As and when it again come in contact, it jumps again. The demo is shown below:
[image]
I want robot to stay in continuous contact of this surface even when the table is moving upside down (say +/- 5cm). Please guide me what I am doing wrong here.
Thanks

Steven89Liu, 2020-08-05:
Could you share your demo with me? thanks.

ma-rv, 2020-08-08:
Hi @deepakraina99,

I would like to ask if you could share your impedance control implementation? I am trying to use cartesian impedance control for insertion, but somehow my implementation is not working properly. I would be really grateful!

Thank you!

closed 2021-04-26 by erwincoumans