bulletphysics/bullet3 #2975: Running multiple threaded pybullet envs in parallel

fgolemo, 2020-08-08:
Hey Erwin,

In another issue, you said, "Each PyBullet DIRECT connection runs independent" (https://github.com/bulletphysics/bullet3/issues/1925). However, if I just use multithreading and `p.connect(p.DIRECT)` this doesn't work 4/10 times. So am I right to assume that `p.connect(p.DIRECT)` and `p = bc.BulletClient(connection_mode=pybullet.DIRECT)` do different things?

Here's a minimal example that illustrates the problem:

```python
import threading
import time
import numpy as np

def make_sim():
    import pybullet as p
    import pybullet_data

    p.connect(p.DIRECT)
    p.setAdditionalSearchPath(pybullet_data.getDataPath())
    plane = p.loadURDF("plane.urdf")
    sphereRadius = 0.05
    colSphereId = p.createCollisionShape(p.GEOM_SPHERE, radius=sphereRadius)
    visualShapeId = p.createVisualShape(shapeType=p.GEOM_SPHERE, rgbaColor=[1, 0, 1, 1], radius=sphereRadius)
    sphere = p.createMultiBody(0.1, colSphereId, visualShapeId, basePosition=[0, 0, sphereRadius + 0.00001])
    p.setGravity(0, 0, -10)
    p.setRealTimeSimulation(0)
    p.applyExternalForce(sphere, -1, [100, 0, 0], posObj=[0, 0, 0], flags=p.LINK_FRAME)
    positions = []
    for _ in range(240):
        p.stepSimulation()
        positions.append(list(p.getBasePositionAndOrientation(sphere)[0]))

    return np.array(positions)

def worker(inst):
    pos = make_sim()
    np.savez(f"inst-{inst}.npz", positions=pos)

w1 = threading.Thread(target=worker, kwargs={"inst":0})
w2 = threading.Thread(target=worker, kwargs={"inst":1})
w3 = threading.Thread(target=worker, kwargs={"inst":2})

w1.start()
w2.start()
w3.start()

w1.join()
w2.join()
w3.join()

import matplotlib.pyplot as plt

x = np.arange(0,240)
data = []
for i in range(3):
    trajectory = np.load(f"inst-{i}.npz")["positions"]
    plt.plot(x, trajectory[:,0])

plt.show()
```

If I run this multiple times, sometimes it works (all trajectories overlap) but sometimes it glitches out and I get these:

[image]
[image]
[image]
[image]

If I may make a recommendation, I think the documentation should be a bit more clear that the default way of running pybullet that's in the doc and most non-gym examples is not thread-safe. (Unless there's something fundamentally wrong with my example above, of course).

erwincoumans, 2020-08-08:
You need to use pybullet_utils.BulletClient
~~~~~
p = bc.BulletClient(connection_mode=pybullet.DIRECT) 
~~~~~

See also the default Bullet Gym environments, that are thread safe and can (and do) run in parallel.

erwincoumans, 2020-08-08:
Thanks Florian. Yes, you are correct. You need to use pybullet_utils.BulletClient for thread safety.
BulletClient will automatically create a new 'client_id' for each connection, and pass that client_id explicitly (under the hood) for each command. That way, the simulations are fully separate. Note that you have to use DIRECT for this, not GUI.
~~~~~
p = bc.BulletClient(connection_mode=pybullet.DIRECT) 
~~~~~

See also the default Bullet Gym environments, that are thread safe and can (and do) run in parallel.

https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_envs/env_bases.py#L51

fgolemo, 2020-08-09:
Okay, thanks.

closed 2020-08-09 by fgolemo