bulletphysics/bullet3 #2982: SaveState and Restore State Non Determinism

ossamaAhmed, 2020-08-11:
Hi,

It seems to me that saveState and restoreState results in a non deterministic behavior sometimes. Please check the script below for reproducing the bug with the KukaGymEnv.

```import numpy as np
from pybullet_envs.bullet.kukaGymEnv import KukaGymEnv

def example():
    env = KukaGymEnv(renders=False,
                     isDiscrete=False)  # operates at 240 HZ
    for j in range(100):
        print("try number ", j)
        actions = [env.action_space.sample() for _ in range(200)]
        env.reset()
        observations_1 = []
        rewards_1 = []
        for i in range(200):
            observations, rewards, _, _ = env.step(actions[i])
            if i == 100:
                state = env._p.saveState()
                print("contacts when saving are ", len(env._p.getContactPoints()))
            observations_1.append(observations)
            rewards_1.append(rewards)
        env._p.restoreState(state)
        print("contacts when restoring state are ", len(env._p.getContactPoints()))
        for i in range(101, 200):
            observations, rewards, _, _ = env.step(actions[i])
            if not np.array_equal(observations_1[i], observations):
                print("step", i)
                print(observations_1[i] - observations)
            assert np.array_equal(observations_1[i] , observations)
    env.close()

if __name__ == '__main__':
    example()
```

Another question I have, what is the recommended way of copying full states across different pybullet instances. SaveState doesnt seem to work across different pybullet instances. The reason I am asking about this is to use it for planning.

seann999, 2020-10-05:
Bump

I'm trying to save grasping demonstration trajectories with saveState and inputs but can't replay them exactly.

erwincoumans, 2020-10-05:
The environment needs to be setup for determinism, using saveState/restoreState and also
~~~~~~~~~~~~~~~
 p.setPhysicsEngineParameter(deterministicOverlappingPairs=1)
~~~~~~~~~~~~~~~

See this working example:
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/unittests/saveRestoreStateTest.py

For different instances, you will have to saveState to disk instead of memory.

closed 2020-10-05 by erwincoumans