bulletphysics/bullet3 #2989: ERROR: failed building pip wheel for pybullet

taiya, 2020-08-14:
Getting errors when installing pybullet (on macOS) within the blender REPL:
```
brew cask install blender
alias bpip='/usr/local/Caskroom/blender/2.83.1/Blender.app/Contents/Resources/2.83/python/bin/pip3'
bpip install pybullet

[...]

 2 warnings generated.
  4 warnings generated.
  18 warnings generated.
  ----------------------------------------
  ERROR: Failed building wheel for pybullet
  Running setup.py clean for pybullet
Failed to build pybullet
```

Attached error log:
[error.txt](https://github.com/bulletphysics/bullet3/files/5075282/error.txt)

erwincoumans, 2020-08-14:
Hi, I've never heard of Blender REPL, but pip install should work with regular python installations.
~~~~~~~~~~~~~
  examples/pybullet/pybullet.c:40:10: fatal error: 'Python.h' file not found
  #include <Python.h>
~~~~~~~~~~~~~
Can you ask the Blender REPL creators about where the Python.h header is?

taiya, 2020-08-14:
Hey Erwin, blender "ships" its own internal version of blender. I can install w/o problems in an hermetic virtualenv python. I thought to create a tracking bug to make sure somebody can find the solution while I search for one :)

taiya, 2020-08-14:
Ok, I resolved the problem.
First, I force reinstalled pybullet on the local virtualenv python3 install:
```
(python3) ~/dev/kubric: pip install --upgrade --force-reinstall pybullet
Processing /Users/atagliasacchi/Library/Caches/pip/wheels/60/d1/2c/c1ae036303601e0c073ab177b0e219fee34d157ee2be2f05a8/pybullet-2.8.7-cp37-cp37m-macosx_10_15_x86_64.whl
Installing collected packages: pybullet
  Attempting uninstall: pybullet
    Found existing installation: pybullet 2.8.7
    Uninstalling pybullet-2.8.7:
      Successfully uninstalled pybullet-2.8.7
Successfully installed pybullet-2.8.7
```

This must have caused dependencies and INCLUDEPATH to be updated, because now the REPL pip3 was able to install it (note it's uninstalling as the previous install already succeeded... I did not save my history):
```
(python3) ~/dev/kubric: bpip install --upgrade --force-reinstall pybullet
Processing /Users/atagliasacchi/Library/Caches/pip/wheels/60/d1/2c/c1ae036303601e0c073ab177b0e219fee34d157ee2be2f05a8/pybullet-2.8.7-cp37-cp37m-macosx_10_15_x86_64.whl
Installing collected packages: pybullet
  Attempting uninstall: pybullet
    Found existing installation: pybullet 2.8.7
    Uninstalling pybullet-2.8.7:
      Successfully uninstalled pybullet-2.8.7
Successfully installed pybullet-2.8.7
```

taiya, 2020-08-14:
@Qwlouse if you can confirm this resolves it for you, I'll close this issue.

Qwlouse, 2020-08-14:
Cannot reproduce currently. Since the last Xcode update I get a bunch of different errors when installing pybullet. They all look like this:
```
examples/ThirdPartyLibs/zlib/gzread.c:31:9: error: implicit declaration of function 'read' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                  ret = read(state->fd, buf + *have, len - *have);
```

But I am pretty sure that your problem was due to some include paths missing, because the blender internal python comes without header files. So you would have to somehow tell it to use the headers from your system python. 
You found a neat way around that problem by installing `pybullet` in a virtualenv where the paths were set correctly, and Blender-python then noticed a cached version of the built `pybullet` package and installed that instead of building it from scratch.

taiya, 2020-08-14:
Yes, what you described is exactly what I think happened.
PyBullet "polluted" the ENV path, and left the modified INCLUDEPATH variables updated with the right  content.
For now this is a simple enough workaround, not elegant by any means...

erwincoumans, 2020-08-14:
> I get a bunch of different errors

Are you installing using 
~~~~~~~~~~~~~
pip3 install pybullet
~~~~~~~~~~~~~
Where is -Werror defined?

bhanu2710, 2021-01-28:
it is giving my very big error

erwincoumans, 2021-01-28:
you need a valid python installation with python.h and a working c++ compiler. custom environments are left as an exercise for the reader, good luck!

closed 2021-01-28 by erwincoumans