bulletphysics/bullet3 #2990: Pybullet URDF Joint Issues

1234shel, 2020-08-15:
I am loading my URDF file into pybullet. After loading, when I use getJointInfo to retrieve link name and parent id, they do match the parent-child combinations in the Joints within the URDF file. Am I missing something here?

I have used flags=p.URDF_MAINTAIN_LINK_ORDER when loading the file.

erwincoumans, 2020-08-15:
No repro with urdf file means we cannot do anything. Also, it is ok if they match.

closed 2020-08-15 by erwincoumans

1234shel, 2020-08-16:
Hi,

I have attached the files. It would be great if you could have a look. I just want to know why the parent and child links are not the same as they are defined in the Urdf file.

On a related note, I had to turn off p.URDF_MAINTAIN_LINK_ORDER. This was causing some elements of the Urdf file to go through the floor. I think I have seen other posts reporting this same issue as well.

[robo_files.zip](https://github.com/bulletphysics/bullet3/files/5079940/robo_files.zip)