bulletphysics/bullet3 #2996: The function getContactPoints apparent strange behaviour where it can find contact points with some .sdf but not another similarly designed .sdf

yungchuenng, 2020-08-19:
Hi!

I am trying to acquire the **contact distance and normal forces data from the contact points of my simulated robot model with objects in the surroundings**. In my testing of the function getContactPoints of a body, i.e. Body_A, in simulation

> link_contact_state = self.pb.getContactPoints(linkIndexA=self.contact_sensor_link_name_index_dic[link_name])

**I find that when the body comes into contact (collision) with some objects in the environment, it is able to reflect the contact points and the required relevant data, but for some other objects in the environment, it does not show any contact points at all**. I singled out two similarly designed .sdf models that were imported by the .world,
A construction cone (body A is able to acquire contact points with this object and its duplicates):
```
<?xml version="1.0"?>
<sdf version="1.6">
  <model name="Construction Cone">
    <static>1</static>
    <pose frame=''>-2 -2 0 0 0 0</pose> 
    <link name="link">
      <inertial>
        <mass>1.0</mass>
        <inertia>
          <ixx>1.0</ixx>
          <ixy>0.0</ixy>
          <ixz>0.0</ixz>
          <iyy>1.0</iyy>
          <iyz>0.0</iyz>
          <izz>1.0</izz>
        </inertia>
      </inertial>
      <collision name="collision">
        <geometry>
          <!--mesh>
            <scale>10 10 10</scale>
            <uri>model://construction_cone/meshes/construction_cone.dae</uri>
          </mesh-->
          <box> <!--commented out mesh and added box to test against the other object if makes any difference for collision 
          detection-->
            <size>0.906 0.05 1.2</size>
          </box>
        </geometry>
      </collision>
      <visual name="visual">
        <geometry>
          <box>
            <size>0.906 0.05 1.2</size>
          </box>
          <!--mesh>
            <scale>10 10 10</scale>
            <uri>package://isr-testbed/construction_cone/meshes/construction_cone.dae</uri>
          </mesh-->
        </geometry>
      </visual>
    </link>
  </model>
</sdf>
```
and a door (body A is **NOT** able to acquire contact points with this object and its duplicates):
```
<?xml version="1.0"?>
<sdf version="1.6">
  <model name="doorrrr"> <!--wall-->
    <static>1</static>
    <pose frame=''>2 2 0 1 -0 0</pose>
    <link name="link">
      <inertial>
        <mass>1.0</mass>
        <inertia>
          <ixx>1.0</ixx>
          <ixy>0.0</ixy>
          <ixz>0.0</ixz>
          <iyy>1.0</iyy>
          <iyz>0.0</iyz>
          <izz>1.0</izz>
        </inertia>
      </inertial>
      <collision name="collision"><!--Door_4_Collision-->
        <geometry>
          <box>
            <size>0.906 0.05 1.2</size>
          </box>
        </geometry>
        <pose frame=''>0 0 0.6 0 -0 0</pose>
      </collision>
      <visual name="visual"><!--Door_4_Visual-->
        <pose frame=''>0 0 0.6 0 -0 0</pose>
        <geometry>
          <box>
            <size>0.906 0.05 1.2</size>
          </box>
        </geometry>
        <!--material>
          <script>
            <uri>file://media/materials/scripts/gazebo.material</uri>
            <name>Gazebo/Grey</name>
          </script>
        </material-->
      </visual>
      <!--pose frame=''>-0.514741 -3.87011 0 0 0 -0.000969</pose-->
    </link>
  </model>
</sdf>
```
I have tried troubleshooting one by one if any of the features affected whether the objects get detected or not, and the following are my apparent findings:

> Changing the value of <static>1</static> to <static>0</static> or vice versa does not affect if either objects are detected. 

> Using a mesh or box geometry does not affect if either objects are detected.

> Adding a material or not does not affect if either objects are detected.

> Having a <pose frame=''>0 0 0 0 0 0</pose> within the <geometry> or <visual> tags or not does not affect if either objects are detected.

> Having the <inertial> data specified in the .sdf or not does not affect if either objects are detected.

Somehow, it feels like the imported stationary walls into the world are the objects which contact points are not being detected by the specified body in the getContactPoints function. 

This is the section of my .world code which imports the models in question (construction cone and door) for reference!
```
    <!-- Door -->
    <include>
        <name>door4</name>
        <uri>file://isr-testbed/Door_4</uri>
        <pose>0 0 0 0 0 0</pose>
    </include>
    <!-- Cone 1 -->
    <include>
        <name>cone1</name>
        <uri>file://isr-testbed/construction_cone</uri>
        <pose>2 -2 0 0 0 0</pose>
    </include>
```
To sum it up, my question is, how is it that some objects may be detected and some objects cannot be detected upon collision? 

Also, I have a side question, is it possible to find the contact points between a body and the ground?

I would greatly appreciate anyone's help in this matter!!

erwincoumans, 2020-08-19:
The full pybullet script is missing. Use getCobtactPoints after stepSimulation, don't use substeps and avoid 'setRealTimeSimulation'

yungchuenng, 2020-08-19:
Hi!

Thank you for your prompt response! Allow me to refer to oscar-lima's open source work on the integration between Pybullet and ROS, pybullet_ros.py here: [https://github.com/oscar-lima/pybullet_ros/blob/melodic/ros/src/pybullet_ros/pybullet_ros.py#L229](url).

I shifted the `self.pb.stepSimulation()` to line 225 before the line: `for task in self.plugins:`, and also commented out line 183 `self.pb.setRealTimeSimulation(0)`. 

However, it does not seem to solve the problem. Here I attach a sample screenshot of my robot in simulation in contact with a cone and a wall (the red plates are the bodies which contact points with objects I intend to determine). 
[image]
On the left in the image is a terminal showing the ROS message that is being published to show the contact distance and normal forces data from the contact points. From here I find that the "back_plate" and "back_left_plate" are able to detect close contact with the green cone, but the "front_plate" and "front_right_plate" are not able to detect contact points information with the yellow wall. 

I actually created my own plugin, with reference to oscar-lima's format, to acquire contact information of relevant bodies with the objects in environment. 

Could there be any other factor leading to this behaviour? Thank you for your time!

erwincoumans, 2020-08-19:
getContactPoints should work just fine. Create a ~10 line pybullet test.py without dependencies, zip it together with assets (urdf, obj etc) in the same directory and share it. Perhaps then someone can run it and look at it.

yungchuenng, 2020-12-13:
Thank you for your suggestion! I ended up not using the external .sdf files which are seemingly unreliable with inconsistent physics interaction when ported into simulation, and used generated geometries through Pybullet create body functions. 

Thanks.

closed 2020-12-13 by yungchuenng