bulletphysics/bullet3 #2998: Parsing Error with URDF

chris0914, 2020-08-19:
Hi everyone,

I already mentioned this error in #2730 and reproduced it, more or less on accident, with pybullet. In the before mentioned Issue I loaded the 'turtlebot.urdf' file within an  QtProject and all the elements of the robot were parsed on the same spot, using the C++ API.

However this time I used pybullet and opened a matplotlib plot before launching the simulation. Below is a MWE to reproduce. The placement in general works, as the cylinder is placed correctly. I reproduced this on one other computer. Both running on Ubuntu 18.04 and I installed matplotlib/pybullet via pip3. The outcome depends on the used turtlebot.urdf file. When using the turtlebot.urdf from pybullet_robots a triangular shape appears, see image below.

    #!/usr/bin/python3
    import pybullet as p
    import matplotlib.pyplot as plt
    from time import sleep

    if __name__ == '__main__':
        
        # Break turtlebot parser
        plt.plot([1,2,3])

        # Simulation initialization
        server_client = p.connect(p.GUI)
        p.resetSimulation()
        p.setRealTimeSimulation(False)
        p.setTimeStep(1./240.0)
        p.setGravity(0, 0, -9.81)
        
        # drop plane and turtlebot
        p.createMultiBody(0, p.createCollisionShape(p.GEOM_PLANE))
        p.createMultiBody(0, p.createCollisionShape(shapeType=p.GEOM_CYLINDER, radius=0.1, height=1.0, collisionFramePosition=[1.0, 1.0, 0.5]))
        turtle = p.loadURDF("data/turtlebot.urdf", [0, 0, 0], p.getQuaternionFromEuler([0, 0, 0]))

        sleep(10.0)
        

[image]

Any idea where to dig?
Cheers Chris

erwincoumans, 2020-08-20:
No idea, running that code look like this here:

[image]

erwincoumans, 2020-08-20:
No idea, it looks normal here, running the same code:
[image]
I'm sure it runs fine in an online Google Colab as well, likely some driver/GPU/Ubuntu installation issue.

chris0914, 2020-08-20:
Hi Erwin,

thanks for the answer. I rebuild a fresh system for Ubuntu 18.04 and Ubuntu 20.04. Running under python3 both showed the same error. However, on the Ubuntu 20.04 system I installed python2.7 and surprisingly everything runs fine!
So I tested that on my main setup, and indeed using python2.7 everything works fine again.
Any idea? As I wouldn't want to go back to python2.

Cheers Chris

erwincoumans, 2020-09-21:
Can't reproduce, and haven't heard a similar issue before. Perhaps a broken graphics driver? Closing for now.

closed 2020-09-21 by erwincoumans