bulletphysics/bullet3 #3005: Best practice to prevent deformable self penetration

huy-ha, 2020-08-24:
**Issue**: The cloth I'm trying to simulate is experiencing some pretty bad self penetration. I understand support for deformable objects are still experimental features, but is there any way I can mitigate this issue?

**Setup**:
- Python 3.7.6
- PyBullet 2.8.7
- Operating System: Ubuntu 18.04

[image]

My simulation timestep is 1/480, my cloth is a plane with 625 vertices, and my `p.loadSoftBody()` parameters are as follows:
```py
self.body_id = p.loadSoftBody(
            fileName=self.cloth_obj_path,
            basePosition=pose[0],
            baseOrientation=pose[1],
            scale=0.2,
            mass=0.5,
            collisionMargin=0.003,
            useNeoHookean=False,
            useBendingSprings=True,
            useFaceContact=True,
            useMassSpring=True,
            useSelfCollision=True,
            springDampingAllDirections=False,
            springElasticStiffness=40,
            springDampingStiffness=0.1,
            frictionCoeff=0.5
        )
```
If `collisionMargin` is any larger, the cloth is very prone to shaking. Decreasing the timestep further doesn't seem to help.

erwincoumans, 2020-08-24:
It is still experimental/unsupported, I'll make it more clear in the quickstart guide (or remote its mention). There is no-one to support the feature, unfortunately.

erwincoumans, 2020-08-25:
Let me ask some others who have experience with the deformables to look at this.

Could you share a full self-contained pybullet example, so it is quicker to reproduce (something we can run out-of-the-box using pybullet). Note you can attach zip files if needed to this tracker.

huy-ha, 2020-08-25:
Absolutely! Here](https://github.com/bulletphysics/bullet3/files/5126587/cloth-penetration-example.zip) is a self-contained directory. Thank you so much!

fuchuyuan, 2020-08-28:
Thanks for reporting this issue!

Increasing the collision margin might cause strong repulsion force for nearby nodes, which causes the instability. The penetration could partly due to rendering, and the cloth in general looks rather jaggy in pybullet. You can try to increase the size of the cloth (say scale=1) to see if it helps.

Like Erwin said, deformable is still experimental. The self-penetration solution is still being improved. One known issue is that once there is penetration, we don't recover from it. Another problem is that once a point or triangle touches the ground, it will ignore repulsion from any other triangles, so you need to be careful manipulating it.

I changed the scale to 1 and hand-manipulated using mouse pick on top of your example. You can see some gap between different parts of the cloth. Hope this helps. 
[image]

[image]

huy-ha, 2020-08-28:
> You can try to increase the size of the cloth (say scale=1) to see if it helps.

I found that if the cloth vertices move relatively slowly compared to their collision margin, self-collision was less likely, which could be why increasing the size of the cloth actually helps? Thanks for the suggestion!

> Another problem is that once a point or triangle touches the ground, it will ignore repulsion from any other triangles, so you need to be careful manipulating it.

Ah, I wasn't aware of this issue. So if one side of the cloth is touching the ground and the other is falling on top of it, would this likely lead to self penetration then?

closed 2021-04-26 by erwincoumans