bulletphysics/bullet3 #3024: Question about math

ssdemajia, 2020-09-01:
Hi, I'm reading the bullet source code, but there is a expression I don't know in `btVoronoiSimplexSolver.cpp`. Why calculate `btScalar vc = d1 * d4 - d3 * d2;` Is a math expression?
```
// Check if P in edge region of AB, if so return projection of P onto AB
	btScalar vc = d1 * d4 - d3 * d2;
	if (vc <= btScalar(0.0) && d1 >= btScalar(0.0) && d3 <= btScalar(0.0))
	{
		btScalar v = d1 / (d1 - d3);
		result.m_closestPointOnSimplex = a + v * ab;
		result.m_usedVertices.usedVertexA = true;
		result.m_usedVertices.usedVertexB = true;
		result.setBarycentricCoordinates(1 - v, v, 0);
		return true;
		//return a + v * ab; // barycentric coordinates (1-v,v,0)
	}
```
Code is in function `closestPtPointTriangle` https://github.com/bulletphysics/bullet3/blob/master/src/BulletCollision/NarrowPhaseCollision/btVoronoiSimplexSolver.cpp

erwincoumans, 2020-09-01:
This is an optimization, as described in the book 'real-time collision detection' by Christer Ericsson, chapter 5.1.5

--------------

// P is outside (or on) AB if the triple scalar product [N PA PB] <= 0
Vector n = Cross(b - a, c - a);
float vc = Dot(n, Cross(a - p, b - p));
// If P outside AB and within feature region of AB, // return projection of P onto AB
if (vc <= 0.0f && snom >= 0.0f && sdenom >= 0.0f)
        return a + snom / (snom + sdenom) * ab;

--------------

closed 2020-09-01 by erwincoumans

ssdemajia, 2020-09-01:
> This is an optimization, as described in the book 'real-time collision detection' by Christer Ericsson, chapter 5.1.5
>
> // P is outside (or on) AB if the triple scalar product [N PA PB] <= 0
> Vector n [...]

Thank you :D