bulletphysics/bullet3 #3033: pybullet.calculateJacobian segfaults when using numpy as input

TimSchneider42, 2020-09-02:
Hi,

I noticed that pybullet.calculateJacobian causes a segmentation fault when one of the inputs is a numpy array. This is of course not a big deal if you know it as you can simply convert all inputs to python lists first, but it took me a while to figure it out.

Here is a minimal example that produces the bug on my system:
```python
import pybullet_data
import pybullet as p
import numpy as np

print("Version: {}".format(p.getAPIVersion()))

p.connect(p.GUI)

p.setAdditionalSearchPath(pybullet_data.getDataPath())
body = p.loadURDF("cartpole.urdf")

pos = np.zeros(2)  # If this is replaced by [0, 0], it works

print(p.calculateJacobian(body, 1, [0, 0, 0], pos, [0, 0], [0, 0]))
p.disconnect()

```
will produce
```
pybullet build time: May 16 2020 11:29:16
Version: 201911280
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=GeForce GTX 1070/PCIe/SSE2
GL_VERSION=3.3.0 NVIDIA 450.66
GL_SHADING_LANGUAGE_VERSION=3.30 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.3.0 NVIDIA 450.66
Vendor = NVIDIA Corporation
Renderer = GeForce GTX 1070/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started

Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)
```

I noticed the same behavior for some other functions as well (I think the inverse kinematic for example), but I am not sure about that.

I am using python 3.8 if that makes any difference.

Best,
Tim

erwincoumans, 2020-09-02:
Indeed, the Python bindings are hand-written and don't expect Numpy for most API calls. Should put that in the user manual.

TimSchneider42, 2020-09-02:
My pybullet projects all involve numpy to some degree and I can imagine that this holds true for many other people as well. So I think it would be great to have support for that if it is not too much work.

erwincoumans, 2020-09-02:
It is likely a huge amount of work, so I wouldn't expect it:
It all happens in this one file. We enabled Numpy for 1 API in pybullet_rayTestBatch:
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/pybullet.c#L6681
So probably closing it for now, we don't have resources to work on this.
If anyone has time, re-open it or create a new issue.

closed 2020-09-02 by erwincoumans