bulletphysics/bullet3 #3034: trouble in installing bullet

WangMengqi32C, 2020-09-02:
Dear all,

I have a weird error during installing the bullet.
 C:\Temp\pip-install-40bxpaah\pybullet\examples\ThirdPartyLibs\zlib\gzguts.h(23): fatal error C1083: Cannot open include file: 'unistd.h': No such file or directory                                                                             error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.27.29110\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2                                                                          ----------------------------------------                                                                            ERROR: Command errored out with exit status 1: 'C:\ProgramData\Anaconda3\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Temp\\pip-install-40bxpaah\\pybullet\\setup.py'"'"'; __file__='"'"'C:\\Temp\\pip-install-40bxpaah\\pybullet\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Temp\pip-record-92tvjfcy\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\ProgramData\Anaconda3\Include\pybullet' Check the logs for full command output.
[image]

I'm using windows 10, and installing with virtual studio 2019 with pip.

Thank you in advance!
Best regards,
Mengqi

erwincoumans, 2020-09-03:
The Windows build was broken due to this PR:
https://github.com/bulletphysics/bullet3/pull/3012
It is fixed now, please try

pip install pybullet --upgrade

closed 2020-09-03 by erwincoumans

WangMengqi32C, 2020-09-03:
thank you a lot! @erwincoumans