bulletphysics/bullet3 #3045: resetDebugVisualizerCamera very slow now

222464, 2020-09-09:
Hello,

I have used PyBullet on and off since its release. I noticed that the latest version(s) of PyBullet made my projects run really slow.  I narrowed it down to the resetDebugVisualizerCamera function (I use it to implement a follow camera). With that line removed, it is as fast as it used to be. However, with it I get like 20fps.

I tried narrowing it down further by commenting out a part of resetDebugVisualizerCamera's source, it doesn't seem to be the connection check. So, it must be in one of the b3 calls. That's as far as I got, as I don't know the API well enough.

Any ideas?

erwincoumans, 2020-09-09:
Thanks for the report. I have never seen the performance impact of this resetDebugVisualizerCamera's.
0) Are you using the latest PyBullet version 2.9.5?
1) What GPU and operating system do you use?
2) Did you try profiling it? In the GUI window, hold 'p' key for 1 second, then use Google Chrome to open the timings.json file. It usually is stored in the /tmp folder on Linux and Mac OSX, or in the current working dir for Windows.
3) A small reproduction case is important too, it could be the assets etc.

222464, 2020-09-09:
Did you just change something? Since I upgraded and now it works fine again!

So here is what I did today:
- installed via pip (my computer was wiped, so I had to reinstall fresh). The issue occurred here.
- installed from the repository directly to make sure I had the latest. The issue still occurred, but for some reason the version number said 2.9.4?
- upgraded on pip again, now it says 2.9.6, and it works again!

In case you still need it:
1. Manjaro Linux, Vega 64.
5. Once per simulation step

erwincoumans, 2020-09-09:
Yes, I just fixed it in 2.9.6.
You may still want to disable wait on v-sync. For example, using nvidia-settings (if you have a NVIDIA GPU), AMD and Intel and other GPUs have similar setting.

222464, 2020-09-09:
Excellent, thanks!

closed 2020-09-09 by 222464

erwincoumans, 2020-09-09:
By the way, did you check out this MPC example?
https://www.youtube.com/watch?v=KPFJdC8Tcm0
You can try it out, the instructions are in the video description. Latest PyBullet has mipmapping enabled, looks much smoother.

222464, 2020-09-09:
Just tried that example, seems like an interesting task. I will play around with it!

I have contributed to some other simulations, but I haven't created any of my own with PyBullet. I fleshed out a simulation for the Stanford Pupper robot to make experimenting with that easier. I didn't use it for sim2real, though - I work on stuff that works IRL directly. It was still handy to debug the RL algorithms, though.

Here is a video if what came of that, in case it interests you: [https://www.youtube.com/watch?v=Zl6Rfb3OQoY](https://www.youtube.com/watch?v=Zl6Rfb3OQoY)

Also, I have used regular C++ bullet for a reinforcement learning environment a while ago: [https://github.com/222464/PGE](https://github.com/222464/PGE). I haven't touched that engine in a while, but perhaps I should revive it. It has modern 3D graphics (screen space reflections / global illumination, deferred shading, parallax occlusion mapping, CSM, surface net voxel terrain, etc.), and bullet does all the physics.

erwincoumans, 2020-09-09:
Thanks for sharing, the PGE project looks cool!

For PyBullet, I kept the OpenGL 3 features simpler than PGE (just shadow mapping, single light, a few GLSL shaders), but it takes effort to maintain compatibility with Windows, Linux and Mac OSX and EGL.