bulletphysics/bullet3 #3046: Can't get NodeID of Softbody (PyBullet)

Simon-Steinmann, 2020-09-10:
Hi, 
After 2days of researching, source code inspection and hours of github issues reading, I'm at my wit's end.
My goal is, to simulate a robot with a suctiono gripper, to pic up textiles. I got so far as to add and remove softBodyAnchors, however, the issue is, that I have to know the nodeID of where to attach the anchor. I tried getting that ID by using 
`p.rayTestBatch(pos1, pos2)`, but it only returns the object ID and `-1 `as the nodeID. 

I managed to get the mouse picking to work on the textile, making me wonder, whether something like this could be used. Applying forces to the softBody, where a rayTest intersects it. is this possible somehow?

I would really appreciate it, if someone could answer any of these questions, propose a different solution or point me in the right direction.

erwincoumans, 2020-09-11:
I just spend a few days improving the deformables/soft body, and also allowed to get the mesh data of the simulation mesh.
[image]
Will report when the example/pull request is there.

Simon-Steinmann, 2020-09-12:
Thank you so much. This is highly appreciated. If I can do anything to help or test, let me know. My c++ skills are limited (can read it a bit, but not proficient coder). But I'm good with python. Will this allow one to get the nodeID of a softbody?

closed 2021-04-26 by erwincoumans