bulletphysics/bullet3 #3049: Resting contacts problem

mcsoeren, 2020-09-11:
Hello everyone,
I have been working with PyBullet for some time as part of a term paper and have encountered the following problem:
If I place a smaller cube (0.5x0.5x0.5m, mass=1) on top of a larger one (1x1x1m, mass=0) and just add gravity, I observe that the small cube moves. Over a simulated period of about 10 hours, it is about 0.25m in y-direction and 0.05m in x-direction. In addition it rotates a little bit.
This movement should actually be prevented by frictional forces. Is it possible that it has something to do with microcollisions? And if so, shouldn't they occur randomly in the plane? Instead, the cube always moves in a certain direction. I can influence this by changing the starting orientation. 
Does anyone have an idea how to solve the problem?

This is the code I used:
```
import pybullet as p
import pybullet_data
import math

simulated_hours = 10
simulated_seconds = simulated_hours*3600
simulated_steps = simulated_seconds*240

gravity = 9.81
small_box_size = [0.5, 0.5, 0.5]
big_box_size = [1, 1, 1]
small_box_start_pos = [0, 0, 0]
small_box_start_ori = [0, 0, math.pi/2]
big_box_start_pos = [0, 0, -((big_box_size[2]/2)+(small_box_size[2]/2))]
physicsClient = p.connect(p.GUI)
p.setAdditionalSearchPath(pybullet_data.getDataPath())
p.setGravity(0, 0, -gravity)
small_box = p.createCollisionShape(shapeType=p.GEOM_BOX, halfExtents=[small_box_size[0]/2, small_box_size[1]/2,
                                                                      small_box_size[2]/2])
testobj_ID = p.createMultiBody(baseMass=1, baseCollisionShapeIndex=small_box,
                               basePosition=small_box_start_pos,
                               baseOrientation=p.getQuaternionFromEuler(small_box_start_ori))
big_box = p.createCollisionShape(shapeType=p.GEOM_BOX, halfExtents=[big_box_size[0]/2, big_box_size[1]/2,
                                                                      big_box_size[2]/2])
base_ID = p.createMultiBody(baseMass=0, baseCollisionShapeIndex=big_box, basePosition=big_box_start_pos)

p.changeDynamics(testobj_ID, -1, lateralFriction=1)
p.changeDynamics(testobj_ID, 0, lateralFriction=1)
p.changeDynamics(base_ID, -1, lateralFriction=1)
p.changeDynamics(base_ID, 0, lateralFriction=1)
step = 0
steps = []
seconds = []
hours = []
x_pos = []
y_pos = []
z_pos = []
x_vel = []
y_vel = []
z_vel = []

while step < simulated_steps:
    steps.append(step)
    seconds.append(step/240)
    hours.append(step/(240*3600))
    p.stepSimulation()
    x_pos.append(p.getBasePositionAndOrientation(testobj_ID)[0][0])
    y_pos.append(p.getBasePositionAndOrientation(testobj_ID)[0][1])
    z_pos.append(p.getBasePositionAndOrientation(testobj_ID)[0][2])
    x_vel.append(p.getBaseVelocity(testobj_ID)[0][0])
    y_vel.append(p.getBaseVelocity(testobj_ID)[0][1])
    z_vel.append(p.getBaseVelocity(testobj_ID)[0][2])
    step += 1

p.disconnect()
```

erwincoumans, 2020-09-11:
There is some small drift that accumulates over time. There are a few options:

0) enable friction anchors, that explicitly deal with this frictional drift in contact
~~~~~~~~~~~~
pybullet.changeDynamics(objectUniqueId, linkIndex, frictionAnchor=1)
~~~~~~~~~~~~
1) enable sleeping/deactivation
~~~~~~~~~~~~
pybullet.loadURDF(fileName, flags=pybullet.URDF_ENABLE_SLEEPING)
~~~~~~~~~~~~
2) increase the number of solver iterations, this will reduce drift but not fully eliminate it (default iterations = 50).
For example:
~~~~~~~~~~~~
pybullet.setPhysicsEngineParameter(numSolverIterations=250)
~~~~~~~~~~~~
3) descrease the simulation time step (from 1./240. to 1./1000.)
~~~~~~~~~~~~
pybullet.setTimeStep(1./1000.)
~~~~~~~~~~~~
4) use a different constraint solver
~~~~~~~~~~~~
p.setPhysicsEngineParameter(constraintSolverType=p.CONSTRAINT_SOLVER_LCP_DANTZIG,
                            globalCFM=0.000001)
~~~~~~~~~~~~

I would recommend start with friction anchors and/or sleeping.

mcsoeren, 2020-09-15:
Thank you very much!
The friction anchor worked and kept the cube in its place. Could you explain shortly what these anchors do? Are they also applicable in dynamic situations? I haven't found any information online.

erwincoumans, 2020-09-21:
Friction anchors keep track of (and counter-act) numerical drift in static friction.
Yes, they are also applicable in dynamic sitations, we use them for our locomotion research as well.

closed 2020-09-21 by erwincoumans