bulletphysics/bullet3 #3050: "App_ExampleBrow" received signal SIGSEGV, Segmentation fault.

slipch, 2020-09-11:
Segmentation fault running an example with the yesterday (10 Sep 2020) master. 
App_ExampleBrow keeps crashing on start.

Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=GeForce GTX 750/PCIe/SSE2
GL_VERSION=3.3.0 NVIDIA 390.138
GL_SHADING_LANGUAGE_VERSION=3.30 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.3.0 NVIDIA 390.138
Vendor = NVIDIA Corporation
Renderer = GeForce GTX 750/PCIe/SSE2
b3Printf: Selected demo: Grasp Deformable Cloth
startThreads creating 1 threads.
starting thread 0
[New Thread 0x7fffe666c700 (LWP 30445)]
started thread 0 
MotionThreadFunc thread started
b3Printf: robotSim connected = 1
b3Printf: numJoints = 8
b3Printf: joint[0].m_jointName=base_joint
b3Printf: joint[1].m_jointName=base_joint_motor
b3Printf: joint[2].m_jointName=motor_left_hinge_joint
b3Printf: joint[3].m_jointName=motor_right_hinge_joint
b3Printf: joint[4].m_jointName=gripper_left_hinge_joint
b3Printf: joint[5].m_jointName=gripper_finger_left
b3Printf: joint[6].m_jointName=gripper_right_hinge_joint
b3Printf: joint[7].m_jointName=gripper_finger_right
b3Printf: b3Warning[/home/slipch/develop/bullet/bullet3/examples/SharedMemory/../Importers/ImportURDFDemo/UrdfFindMeshFile.h,102]:

b3Printf: : cannot find '../../../../data//home/slipch/develop/bullet/bullet3/build_cmake/examples/ExampleBrowser/App_ExampleBrowser.runfiles/google3/third_party/bullet/data/flat_napkin.obj' in any directory in urdf path

Thread 2 "App_ExampleBrow" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe666c700 (LWP 30445)]
0x00005555556a6e78 in btDeformableMassSpringForce::addScaledElasticForce(double, btAlignedObjectArray<btVector3>&) ()
(gdb) bt
#0  0x00005555556a6e78 in btDeformableMassSpringForce::addScaledElasticForce(double, btAlignedObjectArray<btVector3>&) ()
#1  0x00007ffff780bc96 in btDeformableBackwardEulerObjective::applyExplicitForce(btAlignedObjectArray<btVector3>&) ()
   from /home/slipch/develop/bullet/bullet3/build_cmake/src/BulletSoftBody/libBulletSoftBody.so.2.89
#2  0x00007ffff7813e6c in btDeformableBodySolver::predictMotion(double) ()
   from /home/slipch/develop/bullet/bullet3/build_cmake/src/BulletSoftBody/libBulletSoftBody.so.2.89
#3  0x00007ffff7828b4b in btDeformableMultiBodyDynamicsWorld::internalSingleStepSimulation(double) ()
   from /home/slipch/develop/bullet/bullet3/build_cmake/src/BulletSoftBody/libBulletSoftBody.so.2.89
#4  0x00007ffff7820b5c in btDeformableMultiBodyDynamicsWorld::stepSimulation(double, int, double) ()
   from /home/slipch/develop/bullet/bullet3/build_cmake/src/BulletSoftBody/libBulletSoftBody.so.2.89
#5  0x000055555568012b in PhysicsServerCommandProcessor::processForwardDynamicsCommand(SharedMemoryCommand const&, SharedMemoryStatus&, char*, int) ()
#6  0x00005555556a302e in PhysicsServerCommandProcessor::processCommand(SharedMemoryCommand const&, SharedMemoryStatus&, char*, int) ()
#7  0x000055555565f598 in PhysicsServerSharedMemory::processClientCommands() ()
#8  0x0000555555649b9c in MotionThreadFunc(void*, void*) ()
---Type <return> to continue, or q <return> to quit---

erwincoumans, 2020-09-11:
It reports why it terminates:
b3Printf: : cannot find '../../../../data//home/slipch/develop/bullet/bullet3/build_cmake/examples/ExampleBrowser/App_ExampleBrowser.runfiles/google3/third_party/bullet/data/flat_napkin.obj' in any directory in urdf path

erwincoumans, 2020-09-11:
It shouldn't 'crash' though. If you have a small reproduction case, please share.

slipch, 2020-09-11:
ok
Why it is not there?
I didn't remove it.
Just cloned the repo
How to fix ExampleBrowser

slipch, 2020-09-11:
I went through examples one by one until one of them crashed. Now ExampleBrowser always crashing on the start probably because it saves the current example and it is most likely broken

erwincoumans, 2020-09-11:
It must be some test file someone left over in the 'experiments' example.
Delete the 0_Bullet3Demo.txt file, and also comment-out
~~~~~~~~~~~~~~~~
//m_robotSim.loadDeformableBody("flat_napkin.obj", args);
~~~~~~~~~~~~~~~~
in the Bullet/examples/RoboticsLearning/GripperGraspExample.cpp

Thanks for the report, will fix it.

closed 2020-09-24 by erwincoumans