bulletphysics/bullet3 #3052: Heightfield invisible to synthetic camera when in direct mode

ac-93, 2020-09-12:
Hi, 

Apologies if I'm missing something obvious.

When viewing a heightfield body from simulated camera in GUI mode everything is visible, when in direct mode the heightfield is not there. Other types of objects don't seem to have this problem though.

I've attached a script to demo this issue, any idea why this is?

Version details:
Ubuntu 18.04
PyBullet version 2.9.6
Nvidia 2080ti

Thanks,
Alex
[heightfield_test.zip](https://github.com/bulletphysics/bullet3/files/5212474/heightfield_test.zip)

ac-93, 2020-09-16:
Update, managed to work around this problem by creating temporary .obj files with meshio and a temporary urdf file, then loading with p.loadURDF(). This is similar to the possibly related issue #2777, if anybody has the same problem here's a [gist](https://gist.github.com/ac-93/fdc8fb48dac312dae0bd70da6b66a356) for the workaround.

This should be good enough for my use case, happy to close if there's no better solutions being worked on.

erwincoumans, 2020-09-16:
I recently implemented textured rendering of deformables, so it should be probably only a few lines to re-use the same code-path for heightfields. That way, you can also animate the heightfield. 
See https://www.youtube.com/watch?v=_EOvLbEEZHk
and https://www.youtube.com/watch?v=o4JMoVMu3sI
enably by setting 'update' to True here:
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/heightfield.py

ac-93, 2020-09-18:
Hi Erwin,

The ability to update heightfields would be handy for me as currently I'm resetting and reloading which is quite slow.

Would you be able to point me in the right direction for code re-use? I'm not sure my c++ is up to scratch but I'd like to take a look.

erwincoumans, 2020-09-20:
With many vertices, the pure software renderer, TinyRenderer, is very slow for a large heightfield (like 1 second per frame).
I'll create a PR soon.

erwincoumans, 2020-09-20:
See  https://github.com/bulletphysics/bullet3/pull/3072

erwincoumans, 2020-09-21:
It is fixed now.

closed 2020-09-21 by erwincoumans

ac-93, 2020-09-22:
Thanks Erwin, this works great! really appreciate the help!