bulletphysics/bullet3 #3062: Meaning of kpOrg and kdOrg in humanoid_stable_pd

tfederico, 2020-09-15:
Hi,

I am trying to adapt deepmimic in pybullet so that it trains the upper part of the body (chest + arms).

I encountered the `_kpOrg` and `_kdOrg` variables in `humanoid_stable_pd` and I have some questions 

- What do they represent?
- How did you pick their values? If I were to remove any joint, would the values for the remaining joints change?
- Is it intentional the fact that the values of `_kdOrg` are the same values of `_kpOrg` divided by 10?

tfederico, 2020-09-16:
@erwincoumans I have the same question for `build_action_offset`, `build_action_scale`, `build_action_bound_min` and `build_action_bound_max` in `pybullet_deepmimic_env`

tfederico, 2020-09-23:
I figured the meaning of the proportional and derivative term for the PD controller. Could you tell me how you obtained those values?

erwincoumans, 2020-09-24:
All those values are from the original deep mimic repository, so that policies created with deep mimic are compatible.

closed 2020-09-24 by erwincoumans