bulletphysics/bullet3 #3064: Values of state size

tfederico, 2020-09-16:
Hi,

in `pybullet_deepmimic_env` you return a constant value for the `get_state_size` method (197) as the sum of the state pose size, state velocity size and state phase size.

I checked the original DeepMimic implementation and the pose size and velocity size are calculated as follows:

```
int cDeepMimicCharController::GetStatePoseSize() const
{
	return mChar->GetNumBodyParts() * mPosDim - 1; // -1 for root x
}
```

```
int cDeepMimicCharController::GetStateVelSize() const
{
	return mChar->GetNumBodyParts() * mPosDim;
}
```

Hence, I would expect the pose and velocity size to differ by +1/-1. However, in the PyBullet implementation the state pose size is 106 and the state velocity size is 90. In addition, it looks like you calculated the state velocity size as `2*mPosDim*numBodyParts`.

Could you please explain why these values are different and how you calculated them?

erwincoumans, 2020-09-16:
It was a quick port a year ago or so. I think DeepMimic also stacks various state variables together and I think the state is the same size, since we can reuse the same policies.

tfederico, 2020-09-16:
Thanks, I hoped there was a formula to calculate them for bodies with a different number of joints and links

erwincoumans, 2020-09-16:
Yes, there is, but someone needs to spend a little bit of time to clean it up to make it more general. I don't have the cycles for it. I mainly create the project to prepare getting some understanding for my intern Jason Peng last year, which resulted in this project https://xbpeng.github.io/projects/Robotic_Imitation/index.html it has another github repo in the link.

closed 2020-09-16 by erwincoumans