bulletphysics/bullet3 #3069: About pybullet torque tracking problem

ddliu365, 2020-09-18:
Hi, folks,

I am using pybullet for simulation via position control for a humanoid.

I found some joint torque information is missing while inquire by `getJointState`.

See the picture below
[image]

Can anyone help to fix this issue?

erwincoumans, 2020-09-18:
We need a small reproduction case. Did your model have fixed joints in the URDF? Did you skip those? Also, did you enable force-torque sensor for the right joints (skipping the fixed joints)?

ddliu365, 2020-09-18:
Yes,  I have fixed joints in URDF, however, those are not listed in the figure. Would that matter?

while I inquire the jointID, I basically match the name with the joint id. What do you mean by skipping?

I didn't enable force-torque sensor for any joint.

erwincoumans, 2020-09-21:
Closing for now. If you can share a small pybullet reproduction case, we can re-open.

closed 2020-09-21 by erwincoumans