bulletphysics/bullet3 #3073: Availability of pybullet on conda-forge

gonzalocasas, 2020-09-21:
_NOTE: I am sure this has been asked before but I have not found a completely clear explanation on closed issues._

---

I find it very confusing that it is seemingly impossible to *cleanly* install `pybullet` as a conda/conda-forge package.
Here's what I find confusing: 
1. conda-forge has a `bullet` package and installing it actually installs `pybullet`
2. regardless of that, it's been mentioned often that "bullet is not the same as pybullet" (eg [here](https://github.com/bulletphysics/bullet3/issues/2254#issuecomment-495182450)), and while they are not the same, the package installs pybullet, so, I don't understand the clarification.
3. after having installed `conda install bullet`, running `pip install pybullet` overrides the installation instead of finding it as already satisfied requirement (which is what happens with mostly every other package out there).

Finally, the conda-forge package has not been updated in about 10 months, while the pip has had very frequent updates in the same period. Normally, conda-forge bot automatically takes care of creating a PR when a new release is detected, so it's not clear to me what this doesn't happen in this case.

erwincoumans, 2020-09-21:
I'm unfamiliar with conda, but I suppose they need to call the package pybullet?

gonzalocasas, 2020-09-21:
> but I suppose they need to call the package pybullet?

Not necessarily.

> Who is the maintainer of conda bullet?

According to the recipe, it is @xantares 

> What about this package? https://anaconda.org/hcc/pybullet

Packages outside conda/conda-forge are not the same level of support. That one in particular seems pretty unmaintained, last released 1+ year ago. That's why forge is so good, it's tied to pypi/github released to auto trigger.

erwincoumans, 2020-09-21:
I recommend using pip install pybullet, that is most up-to-date, aside from git clone and python setup.py install --user.

gonzalocasas, 2020-09-21:
> I recommend using pip install pybullet, that is most up-to-date, aside from git clone and python setup.py install --user.

yes, the recommendation is clear, but neglecting the conda ecosystem which is widely use in scientific computing, feels like a lost opportunity. It is not difficult to keep a conda-forge recipe constantly up-to-date to the latest `pypi` releases. If @xantares agrees, I can definitely help with that maintenance task.

erwincoumans, 2020-09-22:
It would be great to have an updated pybullet in conda-forge. The regular C++ Bullet SDK is almost a year out-of-date, and pypi/pip releases are almost weekly.

gonzalocasas, 2020-09-22:
I pushed a PR to update the conda-forge package, hopefully @xantares agrees with it and we can have a shiny up-to-date bullet/pybullet package available!
https://github.com/conda-forge/bullet-feedstock/pull/9

wolfv, 2020-09-22:
Actually, conda is much more than just a "Python" library manager, as it also ships compiled `.so` libraries. It's more like `apt`, `dnf` or `yum` in that sense that it's a binary package manager for any programming language (including C++, Python...).

wolfv, 2020-09-22:
@gonzalocasas unfortunately that also means that your current fix is broken :) since it doesn't ship the shared libraries, or header files, anymore. I'll try to take care of this since this package is very important for my robotics work.

wolfv, 2020-09-22:
also @erwincoumans is there a fundamental reason for not tagging releases on github? it seems quite strange to me to have a different version on pypi that is not tagged on github.

erwincoumans, 2020-09-24:
Github releases are primarily about the C++ SDK.
Pypi are the main pybullet releases. Note that the PyBullet version is not the same as the C++ Bullet SDK version, they are different things.

xantares, 2020-09-24:
we would need the library AND the Python bindings, so we're sticking to the github archive which is older as the pypi archive does not provide the library

erwincoumans, 2020-09-24:
I'll mark a new github release this week.

wolfv, 2020-09-24:
But does the pypi archive also ship the source for bullet? Or is the bullet C++ side not changing at all, only the Python bindings?

gonzalocasas, 2020-09-24:
*note: @erwincoumans please correct me if I'm wrong*

It looks to me like pypi's archive for `pybullet` definitely includes the C++ source (according to [the manifest](https://github.com/bulletphysics/bullet3/blob/master/MANIFEST.in), it includes most of `src`, `examples` and `Extras` directories from github). The source code gets compiled on the user's machine during `setup.py`.

Regarding the question of C++ not changing at all: doesn't seem to be the case. There are clearly changes to the C++ code that are being shipped regularly to `pybullet` installs. So I would argue that the pybullet version number should actually also be the bullet c++ version number, because the c++ code shipped is not a match. Ie: `pybullet 3.0.4` is definitely not the same as `bullet 2.98 + python bindings 3.0.4`.

Considering how tightly coupled both these are, I don't quite understand the split version evolution.

erwincoumans, 2020-09-24:
OK, point taken. I tagged the Bullet C++ release today as 3.05
https://github.com/bulletphysics/bullet3/releases/tag/3.05
And will more regularly keep the git tags in sync with PyBullet, so that conda doesn't go out-of-date that much.
Still, using pip3 install pybullet --user --upgrade is the easy/recommended way.

erwincoumans, 2020-09-26:
Can you let me know if/when the conda release is updated to latest 3.05 release? Please check if pybullet works.

wolfv, 2020-09-27:
Are you interested in being a maintainer of the conda-forge package as well? There is a bot which checks GitHub or pypi releases and triggers automatic updates, so most of the time it's just about pressing a green merge button.

erwincoumans, 2020-11-25:
can anybody update the pubullet version on conda?

erwincoumans, 2020-11-30:
@wolfv or @gonzalocasas 
How can we arrange for the latest pubullet version to be included in conda?
3.07 pybullet and C++ is in sync now.

wolfv, 2020-12-01:
I can work on this today!

@erwincoumans do you also want to become a maintainer of bullet on conda-forge?

wolfv, 2020-12-01:
@erwincoumans I have been looking into this for the last hour or so. The main issue is that there are a lot of vendored libraries (Eigen, BussIK, gwen) that we have to get out of the pybullet / the examples before we can release this to conda-forge, otherwise there might be incompatibilities when loading pybullet and other Python libraries that depend on those.
I suspect you don't have an easy way to use out-of-tree-dependencies, right?

wolfv, 2020-12-02:
thanks for the response. yeah, I am not sure what the best way to go is. I have a working build for the C++ part of bullet, but the python part is ... really tricky with all these thirdparty dependencies, and it seems difficult to get them out of the build system.

I'll ask on conda-forge gitter if we should statically link all this or if we don't accept this in conda-forge. 

I know that it's much easier to have such a tight control over everything, but it really hurts the "proper" packaging effort to vendor all these libraries :(

erwincoumans, 2020-12-02:
Well, can conda just output a message to use pip instead, if someone tries to run conda install pybullet? Again, pybullet is not the same as Bullet C++

gonzalocasas, 2020-12-02:
> Well, can conda just output a message to use pip instead, if someone tries to run conda install pybullet?

Not really, afaik.

> Again, pybullet is not the same as Bullet C++. the setup.py contains all instructions how to build PyBullet for Windows, Linux and Mac.

You usually say this, but I keep failing to understand how they are not the same thing. It's the exact same repo. Pybullet is bindings for the exact same C++ code. Right? What do you refer to exactly when you say they are not the same?

erwincoumans, 2020-12-02:
Traditional Bullet C++ is LinearMath, BulletCollision, BulletDynamics and BulletSoftBody. 

All the other code, including example code for URDF loading and rendering is specific to PyBullet. That is all included in setup.py

erwincoumans, 2020-12-02:
Since we are not likely getting anywhere with conda, probably better to closing this with "cannot fix" for now, and recommend to simpky use 

~~~~~~~~~~~
pip install
~~~~~~~~~~~

closed 2020-12-02 by erwincoumans

wolfv, 2020-12-02:
@erwincoumans I understand that there is some frustration, sorry for that. I guess no-one really has the time to untangle the build system, maybe make pybullet standalone (properly depending on bullet-cpp with dynamic (or static) linkage) etc. 

I think if you would want to package for Debian / Fedora etc. you would be met with similar concerns, and conda-forge really is a distribution just like Debian / Fedora (with the difference of being multi-platform).

I'll see if we can add a statically linked pybullet to conda-forge (even though it's not desired). 

We can just execute the same `setup.py`.
It wouldn't even depend on `bullet-cpp` as far as I can tell, since that is also statically linked into `pybullet`, correct?

gonzalocasas, 2020-12-02:
> We can just execute the same setup.py.

that would basically be the same thing I tried a while ago which incorrectly removed the c++ part, right? maybe if we try that on a separate conda package?

erwincoumans, 2020-12-02:
setup.py compiles all sources in a single pybullet.so, so it doesn't depend on external Bullet C++ libraries:

~~~~~~~~~~~~~~~~~~~~~
+["src/btLinearMathAll.cpp"]\
+["src/btBulletCollisionAll.cpp"]\
+["src/btBulletDynamicsAll.cpp"]\
~~~~~~~~~~~~~~~~~~~~~

wolfv, 2020-12-02:
We can use the same conda package with multiple outputs:

Ideally we end up with `bullet-cpp`, `pybullet` and `bullet` (`bullet` would depend on both, the C++ and Python package). Also ideally, pybullet would depend on bullet-cpp, but this won't be necessary since we'll build pybullet as statically linked, standalone package.

gonzalocasas, 2020-12-02:
I tried the multiple output thing but didn't make it work, and considering that `pybullet` cannot be made to depend on (a theoretical) `bullet-cpp` package, wouldn't it be correct also to just create a new one called `pybullet` and do as @erwincoumans says and just use `setup.py` in there?

wolfv, 2020-12-02:
both options are fine. If you get the standalone pybullet to work, I can combine it with my working bullet-cpp output and we can make a joint PR to the current bullet feedstock :)

gonzalocasas, 2020-12-13:
@wolfv sorry the delay; yes, I can do that. Where should I prepare it? Do I create a brand new staged-recipe for pybullet? or do I create a PR for the existing `bullet` package, but we don't merge, instead you'd take it and do the combination with bullet-cpp?

wolfv, 2021-03-23:
@erwincoumans I finally got around to compile bullet-cpp and pybullet in this PR: https://github.com/conda-forge/bullet-feedstock/pull/28

Do you have any feedback?