bulletphysics/bullet3 #3078: Installing a custom plugin with pybullet

ojasjoshi, 2020-09-23:
Is there a a way to use dynamic plugins with pybullet?

If not, what would be the procedure to install the custom static plugin into bullet physics SDK source code? Would it be, 
- add plugin source code into the bullet source code
- register the plugin using registerStaticLinkedPlugin in the PhysicsServerCommandProcessor
- re-compile bullet physics SDK from source

erwincoumans, 2020-09-24:
You could, but it is probably easier to write your own Python program from scratch, without using PyBullet code.

closed 2020-09-24 by erwincoumans

ojasjoshi, 2020-09-24:
The documentation suggests only C/C++ plugins are supported. Is there a way to understand how python plugins would work?

erwincoumans, 2020-09-24:
I meant writing your own c-extension for Python:
https://docs.python.org/3/extending/extending.html
Or use pybind for your own code, not related to PyBullet.