bulletphysics/bullet3 #3080: pybullet compatibility for specific Ubuntu releases

ojasjoshi, 2020-09-24:
The documentation mentions that Linux distrbutions are supported but does not mention info on the compatibility for Ubuntu 18.04, 16.04 and 14.04 specifically. My question:
- Is pybullet officially supported on all three(18,16,14) Ubuntu releases?
- Are there any plans to deprecate compatibility for certain Ubuntu releases in future pybullet releases?

erwincoumans, 2020-09-24:
PyBullet should work on all Ubuntu versions, it is plain C++ code, nothing fancy.

closed 2020-09-24 by erwincoumans

ojasjoshi, 2020-09-24:
thanks for the quick response @erikogenvik.. the same logic would work on MacOS and Windows but clarifying to check if there are any constraints for compatibility for certain MacOS and Windows releases?

erwincoumans, 2020-09-24:
I'm not aware of incompatibilties.