bulletphysics/bullet3 #3084: Environments descirptions

modanesh, 2020-09-28:
Where can I find each environment description? Like the observations and actions range and what each value refers to, maximum reward, and considered solved reward etc.

Steven89Liu, 2020-09-29:
i think you should can get these values from the each env definition from the bullet3/examples/pybullet/gym/pybullet_envs/, but i also didn't find any more detail description.
hope this can help you.

modanesh, 2020-09-29:
I knew about that. There are also some information here: ```examples/pybullet/gym/pybullet_envs/__init__.py```. But I was looking for something more in detail, like [this](https://github.com/openai/gym/blob/master/gym/envs/classic_control/cartpole.py#L28) for CartPole from OpenAI gym.

closed 2021-04-26 by erwincoumans