bulletphysics/bullet3 #3091: How to make a soft body file for loading in pybullet

deepakraina99, 2020-10-04:
I saw some examples where .obj and .vtk files are used for soft bodies. How these files are designed. I used to get .obj and .dae files from standard designing software but those are for rigid bodies. Loading those files (.obj) using p.loadSoftBody gives no effect as a soft body. Can someone guide me regarding this?

Thanks.

yangdeai, 2020-10-04:
I also want to know this too, please!! I want to simulate catheter,blood vessel and so on. Of course, I still need to simulate a robot to control the catheter into the blood vessel pointing position!

HenningLuo, 2020-11-26:
I do not think loading objects in obj format is a good choice, since it only specifies the surface polygons of the object. Unless this is fabric or cloth, objects with volume should collapse when loaded in FEM mode. The vtk files in bullet examples are all generated with gmsh, so maybe that is the right software for modelling soft objects. I am only getting started to build these model as well, please forgive me if I am wrong.

seann999, 2020-12-07:
Bump; I'm trying to use some software called gmsh but I have no idea what I'm doing

deepakraina99, 2020-12-08:
> Bump; I'm trying to use some software called gmsh but I have no idea what I'm doing

Yes, I am also at the same stage. Pybullet should put up some documentation with respect to the modeling of soft bodies.

erwincoumans, 2020-12-08:
see https://github.com/bulletphysics/bullet3/issues/2726

closed 2020-12-08 by erwincoumans