bulletphysics/bullet3 #3093: Why these certain actions can cause the 2DWalker fly?

liusida, 2020-10-05:
I modified some values in walker2d, and put it above the ground initially.
after passing specific actions to the simulation, I saw the body penetrated through the floor. 
if I pass zeros as actions, the robot falls just fine.

Here is the video I recorded. 
[https://youtu.be/HAmIuSOCR14](https://youtu.be/HAmIuSOCR14) 

Here is the reproducible code, please name it `bug.py`.

```Python
import os
import time
from pathlib import Path

import numpy as np
import gym
import pybullet
from pybullet_envs.gym_locomotion_envs import WalkerBaseBulletEnv
from pybullet_envs.robot_locomotors import WalkerBase
import pybullet_data

class MyBug(WalkerBase):
    foot_list = ["foot", "foot_left"]
    def __init__(self, xml):
        WalkerBase.__init__(self, xml, "torso", action_dim=6, obs_dim=22, power=0.40)

    def alive_bonus(self, z, pitch):
        return 0

class MyBugEnv(WalkerBaseBulletEnv):

    def __init__(self, render=False, xml=""):
        self.robot = MyBug(xml)
        WalkerBaseBulletEnv.__init__(self, self.robot, render)

    def _isDone(self):
        return False

if __name__ == "__main__":
    gym.register(id="MyBug-v0", entry_point="bug:MyBugEnv", kwargs={'render': False, 'xml': ""})

eval_env = gym.make(id="MyBug-v0", xml=f"{Path.cwd()}/walker2d_variation.xml", render=True)
obs = eval_env.reset()

# These are the actions to cause the robot fly.
action_list = [
    [-0.2652849, -0.10890063, -0.12446916, -0.49210608, 0.16671477, 0.06189072],
    [-0.4842735, -0.05028579, -0.14372621, -0.6404359, 0.12535053, 0.04066408],
    [-0.8875768, 0.02128746, -0.16854087, -0.7689454, 0.17108788, -0.06770661],
]

ep_len = 0
episodic_reward = 0
exp_id = 1
while True:

    if exp_id%2==0: # in even experiment, everything is fine.
        action = np.zeros_like(action_list[0])
    else: # in odd experiment, these three steps of actions make the robot fly.
        if ep_len < len(action_list):
            action = action_list[ep_len]

    if isinstance(eval_env.action_space, gym.spaces.Box):
        action = np.clip(action, eval_env.action_space.low, eval_env.action_space.high)

    obs, r, done, _ = eval_env.step(action)
    episodic_reward += r

    if done or ep_len > 10: # control the episode length to only 10 steps
        print(f"exp {exp_id}: episode length = {ep_len}, reward = {episodic_reward}.")
        time.sleep(3)
        obs = eval_env.reset()
        ep_len = 0
        episodic_reward = 0
        exp_id += 1

    time.sleep(0.2) # slow down
    ep_len += 1
```

Here is the XML, please save as `walker2d_variation.xml` and put in the same folder.

```xml
<mujoco model="walker2d">
  <compiler angle="degree" coordinate="global" inertiafromgeom="true"/>
  <default>
    <joint armature="0.01" damping=".1" limited="true"/>
    <geom conaffinity="0" condim="3" contype="1" density="1000" friction="0.8 .1 .1" rgba="0.8 0.6 .4 1"/>
  </default>
  <option integrator="RK4" timestep="0.002"/>
  <worldbody>
    <body name="torso">
      <joint armature="0" axis="1 0 0" damping="0" limited="false" name="ignorex" pos="0 0 0" stiffness="0" type="slide"/>
      <joint armature="0" axis="0 0 1" damping="0" limited="false" name="ignorez" pos="0 0 0" ref="1.25" stiffness="0" type="slide"/>
      <joint armature="0" axis="0 1 0" damping="0" limited="false" name="ignorey" pos="0 0 0" stiffness="0" type="hinge"/>
      <geom fromto="0 0 1.7042348341715208 0 0 1.484253450593086" name="torso_geom" size="0.05179482029577558" type="capsule"/>
      <body name="thigh">
        <joint axis="0 -1 0" name="thigh_joint" pos="0 0 1.484253450593086" range="-150 0" type="hinge"/>
        <geom fromto="0 0 1.484253450593086 0 0 0.960545860244181" name="thigh_geom" size="0.050744455602915434" type="capsule"/>
        <body name="leg">
          <joint axis="0 -1 0" name="leg_joint" pos="0 0 0.960545860244181" range="-150 0" type="hinge"/>
          <geom fromto="0 0 0.960545860244181 0 0 0.23824848224877426" name="leg_geom" size="0.04346220162007972" type="capsule"/>
          <body name="foot">
            <joint axis="0 -1 0" name="foot_joint" pos="0 0 0.23824848224877426" range="-45 45" type="hinge"/>
            <geom fromto="0 0 0.23824848224877426 0.2806803830575767 0 0.23824848224877426" name="foot_geom" size="0.03824848224877425" type="capsule"/>
          </body>
        </body>
      </body>
      <!-- copied and then replace thigh->thigh_left, leg->leg_left, foot->foot_right -->
      <body name="thigh_left">
        <joint axis="0 -1 0" name="thigh_left_joint" pos="0 0 1.484253450593086" range="-150 0" type="hinge"/>
        <geom fromto="0 0 1.484253450593086 0 0 0.960545860244181" name="thigh_left_geom" rgba=".7 .3 .6 1" size="0.050744455602915434" type="capsule"/>
        <body name="leg_left">
          <joint axis="0 -1 0" name="leg_left_joint" pos="0 0 0.960545860244181" range="-150 0" type="hinge"/>
          <geom fromto="0 0 0.960545860244181 0 0 0.23824848224877426" name="leg_left_geom" rgba=".7 .3 .6 1" size="0.04346220162007972" type="capsule"/>
          <body name="foot_left">
            <joint axis="0 -1 0" name="foot_left_joint" pos="0 0 0.23824848224877426" range="-45 45" type="hinge"/>
            <geom fromto="0 0 0.23824848224877426 0.2806803830575767 0 0.23824848224877426" name="foot_left_geom" rgba=".7 .3 .6 1" size="0.03824848224877425" type="capsule"/>
          </body>
        </body>
      </body>
    </body>
  </worldbody>
  <actuator>
    <!-- <motor joint="torso_joint" ctrlrange="-100.0 100.0" isctrllimited="true"/>-->
    <motor ctrllimited="true" ctrlrange="-1.0 1.0" gear="100" joint="thigh_joint"/>
    <motor ctrllimited="true" ctrlrange="-1.0 1.0" gear="100" joint="leg_joint"/>
    <motor ctrllimited="true" ctrlrange="-1.0 1.0" gear="100" joint="foot_joint"/>
    <motor ctrllimited="true" ctrlrange="-1.0 1.0" gear="100" joint="thigh_left_joint"/>
    <motor ctrllimited="true" ctrlrange="-1.0 1.0" gear="100" joint="leg_left_joint"/>
    <motor ctrllimited="true" ctrlrange="-1.0 1.0" gear="100" joint="foot_left_joint"/>
    <!-- <motor joint="finger2_rot" ctrlrange="-20.0 20.0" isctrllimited="true"/>-->
  </actuator>
</mujoco>
```

liusida, 2020-10-05:
Oh! I just realize that might be a bug in `loadMJCF()`, because when I changed to shorter float numbers, the 2DWalker stop flies away. For example, I changed `0.23824848224877426` to `0.23`.

liusida, 2020-10-06:
My guess is it might because the function `loadMJCF()` can't handle such long floats. I am now using shorter floats, like `0.23` instead of `0.23824848224877426`, and don't have any walkers fly away yet.

liusida, 2020-10-06:
After training them for a while, they all learned how to fly.
I saw them shaking their "torso" back and forth hardly, I don't know if this weird behavior causes them to fly.

another footage video: https://youtu.be/J9PjVFUM8Ys

erwincoumans, 2020-10-06:
It can happen when hitting a joint limit, exceeding a velocity threshold or apply to large forces.  There is no support available here for this kind of issues.

closed 2020-10-06 by erwincoumans

liusida, 2020-10-08:
Yes, it is a joint limitation problem.

I deleted these three lines, and everything is so far so good. 
```
      <joint armature="0" axis="1 0 0" damping="0" limited="false" name="ignorex" pos="0 0 0" stiffness="0" type="slide"/>
      <joint armature="0" axis="0 0 1" damping="0" limited="false" name="ignorez" pos="0 0 0" ref="1.25" stiffness="0" type="slide"/>
      <joint armature="0" axis="0 1 0" damping="0" limited="false" name="ignorey" pos="0 0 0" stiffness="0" type="hinge"/>
```