bulletphysics/bullet3 #3094: Sticky walls in pybullet

Qwlouse, 2020-10-05:
While generating a simple bouncing balls dataset, I noticed that results differ based on the machine I am running things on. Sometimes the the walls are "sticky" as illustrated here:
[image]

I made sure to run exactly the same versions of `pybullet` (3.0.4) and `numpy` (1.19.2), but the issue persists. 

Here is a minimal example that reproduces the behaviour for a single ball and wall:
```python
# pbdebug.py
import pybullet as pb
client = pb.connect(pb.DIRECT)

# create a (static) wall at x=1
wall_collision_shape = pb.createCollisionShape(pb.GEOM_BOX, halfExtents=(0.1, 1.0, 1.0))
wall = pb.createMultiBody(0., wall_collision_shape, -1, basePosition=(1.1, 0., 0.))

# create a ball with mass 1.0 at the origin that is headed towards the wall
ball_collision_shape = pb.createCollisionShape(pb.GEOM_SPHERE, radius=0.2)
ball = pb.createMultiBody(1.0, ball_collision_shape, -1, basePosition=(0., 0., 0.))
pb.resetBaseVelocity(ball, (1., 0., 0.), (0., 0., 0.))

for current_step in range(400):
  if current_step % 40 == 0:
    position, quaternion = pb.getBasePositionAndOrientation(ball)
    print(position[0])
  pb.stepSimulation()
```

Desired output (using python3.7m shipped with blender on my mac) showing the ball bouncing back from the wall is :
```
0.0
0.1655351293449156
0.3288917548920626
0.4901124652114389
0.6492386581094166
0.8022166259322788
0.794568771815905
0.7869740542480637
0.7794320880648924
0.7719424911154815
```

The problematic output (using python3.6 on my mac / python3.8 in docker below) where the ball sticks to the wall is is: 
```
0.0
0.1655351293449156
0.3288917548920626
0.4901124652114389
0.6492386581094166
0.7985058510278423
0.8000100000000001
0.8000100000000001
0.8000100000000001
0.8000100000000001
```

I created a simple dockerfile that uses Ubuntu 20.04 to help reproduce the issue:
```dockerfile
# pbdebug.Dockerfile
FROM ubuntu:20.04

# --- Install package dependencies
RUN apt-get update --yes  && \
    apt-get install --yes --no-install-recommends \
      build-essential \
      python3-dev \
      python3-pip

# # --- Install Python dependencies
RUN pip3 install --upgrade pip && \
    pip3 install numpy pybullet
```

Build and run using: 
``` bash
docker build -f pbdebug.Dockerfile -t pbdebug:latest .
docker run -it -rm -v"`pwd`:/tmp" -w "/tmp" pbdebug:latest python3 pbdebug.py
```

erwincoumans, 2020-10-05:
At low velocity, the restitution is expected to be zero. That is not 'sticky' but it will not bounce away.
This is due to a threshold to make resting objects a bit more stable. To disable this, use:
~~~~~~~~~~~~~~
pybullet.setPhysicsEngineParameter(restitutionVelocityThreshold=0)
~~~~~~~~~~~~~~

It is not clear why docker behaves different.

erwincoumans, 2020-10-05:
What is the gcc (or clang) version installed on Mac and Ubuntu?

Qwlouse, 2020-10-05:
I tried `pybullet.setPhysicsEngineParameter(restitutionVelocityThreshold=0)` but it does not change the behavior (added it to the code). Also I can reproduce the sticking behaviour on my Debian (gLinux) with python 3.8. 
Here is the build-log for the docker with all relevant version info:
<details>
  <summary>Docker Build log</summary>
 <pre>
Sending build context to Docker daemon  9.216kB
Step 1/3 : FROM ubuntu:20.04
 ---> bb0eaf4eee00
Step 2/3 : RUN apt-get update --yes  &&     apt-get install --yes --no-install-recommends       build-essential       python3-dev       python3-pip
 ---> Running in bb14f26366bb
Get:1 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]
Get:2 http://security.ubuntu.com/ubuntu focal-security InRelease [107 kB]
Get:3 http://archive.ubuntu.com/ubuntu focal-updates InRelease [111 kB]
Get:4 http://archive.ubuntu.com/ubuntu focal-backports InRelease [98.3 kB]
Get:5 http://archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages [177 kB]
Get:6 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1275 kB]
Get:7 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages [11.3 MB]
Get:8 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.4 kB]
Get:9 http://archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [88.7 kB]
Get:10 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [306 kB]
Get:11 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [706 kB]
Get:12 http://archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 Packages [21.6 kB]
Get:13 http://archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [4277 B]
Get:14 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [368 kB]
Get:15 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [114 kB]
Get:16 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [75.9 kB]
Get:17 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 Packages [1169 B]
Fetched 15.1 MB in 2s (8963 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  binutils binutils-common binutils-x86-64-linux-gnu ca-certificates cpp cpp-9
  dpkg-dev g++ g++-9 gcc gcc-9 gcc-9-base libasan5 libatomic1 libbinutils
  libc-dev-bin libc6 libc6-dev libcc1-0 libcrypt-dev libctf-nobfd0 libctf0
  libdpkg-perl libexpat1 libexpat1-dev libgcc-9-dev libgdbm-compat4 libgdbm6
  libgomp1 libisl22 libitm1 liblsan0 libmpc3 libmpdec2 libmpfr6 libperl5.30
  libpython3-dev libpython3-stdlib libpython3.8 libpython3.8-dev
  libpython3.8-minimal libpython3.8-stdlib libquadmath0 libreadline8
  libsqlite3-0 libssl1.1 libstdc++-9-dev libtsan0 libubsan1 linux-libc-dev
  make mime-support openssl patch perl perl-modules-5.30 python-pip-whl
  python3 python3-distutils python3-lib2to3 python3-minimal
  python3-pkg-resources python3-setuptools python3-wheel python3.8
  python3.8-dev python3.8-minimal readline-common xz-utils zlib1g zlib1g-dev
Suggested packages:
  binutils-doc cpp-doc gcc-9-locales debian-keyring g++-multilib
  g++-9-multilib gcc-9-doc gcc-multilib manpages-dev autoconf automake libtool
  flex bison gdb gcc-doc gcc-9-multilib glibc-doc locales gnupg | gnupg2 git
  bzr gdbm-l10n libstdc++-9-doc make-doc ed diffutils-doc perl-doc
  libterm-readline-gnu-perl | libterm-readline-perl-perl libb-debug-perl
  liblocale-codes-perl python3-doc python3-tk python3-venv
  python-setuptools-doc python3.8-venv python3.8-doc binfmt-support
  readline-doc
Recommended packages:
  fakeroot gnupg | gnupg2 libalgorithm-merge-perl manpages manpages-dev
  libfile-fcntllock-perl liblocale-gettext-perl file netbase
The following NEW packages will be installed:
  binutils binutils-common binutils-x86-64-linux-gnu build-essential
  ca-certificates cpp cpp-9 dpkg-dev g++ g++-9 gcc gcc-9 gcc-9-base libasan5
  libatomic1 libbinutils libc-dev-bin libc6-dev libcc1-0 libcrypt-dev
  libctf-nobfd0 libctf0 libdpkg-perl libexpat1 libexpat1-dev libgcc-9-dev
  libgdbm-compat4 libgdbm6 libgomp1 libisl22 libitm1 liblsan0 libmpc3
  libmpdec2 libmpfr6 libperl5.30 libpython3-dev libpython3-stdlib libpython3.8
  libpython3.8-dev libpython3.8-minimal libpython3.8-stdlib libquadmath0
  libreadline8 libsqlite3-0 libssl1.1 libstdc++-9-dev libtsan0 libubsan1
  linux-libc-dev make mime-support openssl patch perl perl-modules-5.30
  python-pip-whl python3 python3-dev python3-distutils python3-lib2to3
  python3-minimal python3-pip python3-pkg-resources python3-setuptools
  python3-wheel python3.8 python3.8-dev python3.8-minimal readline-common
[... 423 lines cut ...]
Running hooks in /etc/ca-certificates/update.d...
done.
Removing intermediate container bb14f26366bb
 ---> 1af812216278
Step 3/3 : RUN pip3 install --upgrade pip &&     pip3 install numpy pybullet
 ---> Running in cd407ea94070
Collecting pip
  Downloading pip-20.2.3-py2.py3-none-any.whl (1.5 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 20.0.2
    Not uninstalling pip at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'pip'. No files were found to uninstall.
Successfully installed pip-20.2.3
Collecting numpy
  Downloading numpy-1.19.2-cp38-cp38-manylinux2010_x86_64.whl (14.5 MB)
Collecting pybullet
  Downloading pybullet-3.0.4-cp38-cp38-manylinux1_x86_64.whl (102.2 MB)
Installing collected packages: numpy, pybullet
Successfully installed numpy-1.19.2 pybullet-3.0.4
Removing intermediate container cd407ea94070
 ---> ed99c0304995
Successfully built ed99c0304995
Successfully tagged pbdebug:latest
 </pre></details>

Qwlouse, 2020-10-05:
Edit: Added `pb.changeDynamics(wall, -1, restitution=1.)` and `pb.changeDynamics(ball, -1, restitution=1.)` (the default is 0 which would lead to no bouncing too). But behaviour is unchanged.

erwincoumans, 2020-10-06:
The restitution issue is an artifact of a some predictive simulation feature (our locomotion and grasping simulation don't use restitution).

Here is an updated script that produces proper restitution. There are some additional settings to avoid gaining energy due to position stabilization. This requires 'maximalCoordinates=True'.

~~~~~~~~~~~~~~~~~~
# pbdebug.py
import pybullet as pb
import time
dt = 1./240.
client = pb.connect(pb.GUI)
pb.setPhysicsEngineParameter(restitutionVelocityThreshold=0, warmStartingFactor=0, useSplitImpulse=True, contactSlop=0)
pb.setPhysicsEngineParameter(enableConeFriction=False)

# create a (static) wall at x=1
wall_collision_shape = pb.createCollisionShape(pb.GEOM_BOX, halfExtents=(0.1, 1.0, 1.0))
wall = pb.createMultiBody(0., wall_collision_shape, -1, basePosition=(1.1, 0., 0.), useMaximalCoordinates=True)
pb.changeDynamics(wall, -1, contactProcessingThreshold=0, restitution=1, linearDamping=0, angularDamping=0)

# create a ball with mass 1.0 at the origin that is headed towards the wall
ball_collision_shape = pb.createCollisionShape(pb.GEOM_SPHERE, radius=0.2)
ball = pb.createMultiBody(1.0, ball_collision_shape, -1, basePosition=(0., 0., 0.), useMaximalCoordinates=True)
pb.resetBaseVelocity(ball, (1., 0., 0.), (0., 0., 0.))
pb.changeDynamics(ball, -1, contactProcessingThreshold=0, restitution=1, linearDamping=0, angularDamping=0)

for current_step in range(40000):
  if current_step % 40 == 0:
    position, quaternion = pb.getBasePositionAndOrientation(ball)
    linvel, angvel = pb.getBaseVelocity(ball)
    print(position[0])
    print("linvel=",linvel)
  time.sleep(dt)
  pb.stepSimulation()
~~~~~~~~~~~~~~~~~~

Qwlouse, 2020-10-06:
These changes resolve my issue. Thanks a lot @erwincoumans, I would have never figured this out myself!

closed 2020-10-06 by Qwlouse