bulletphysics/bullet3 #3103: Pybullet's version of Bullet?!

Simon-Steinmann, 2020-10-08:
Hello, 

what version of Bullet does Pybullet use? Are they always on the same level (3.05 currently) or is Pybullet on Bullet 2.85 (as the readme.md would suggest). Perhaps the documentation should be adjusted, to make this more clear. We are writing a scientific paper, and it is not clear, which version Pybullet uses. Furthermore, are there significant changes between Bullet2 and Bullet3? Most papers about Bullet are based on versions 2.8-2.9. How applicable are these findings today?

erwincoumans, 2020-10-08:
The version is now in sync, currently 3.0.6
(six!) for PyBullet and Bullet. Just refer to PyBullet using this bibtex:
~~~~~~~~~~
@MISC{coumans2020,
author =   {Erwin Coumans and Yunfei Bai},
title =    {PyBullet, a Python module for physics simulation for games, robotics and machine learning},
howpublished = {\url{http://pybullet.org}},
year = {2016--2020}
}
~~~~~~~~~~

The version numbers of PyBullet and Bullet were not related until a few weeks ago. Bullet 2.89 and 3.06 are very similar.

erwincoumans, 2020-10-08:
the version is now in sync, currently 3.0.6
(six!) for PyBullet and Bullet. Just refer to PyBullet using this bibtex

closed 2020-10-08 by erwincoumans

Simon-Steinmann, 2020-10-08:
Thank you very much! I'd suggest updating the readme, to reflect these recent changes :)