bulletphysics/bullet3 #3105: GPU acceleration for pybullet3?

Simon-Steinmann, 2020-10-08:
Hello,
since pybullet is now on bullet3, is gpu compute supported? The documentation is not clear on this. We are working on a scientific paper about a simulation benchmark, comparing different simulation environments (Gazebo, Mujoco, Webots, Pybullet) with a focus on industrial application and reinforcement learning. Currently the gpu is not used at all on pybullet. Can gpu acceleration be enabled? The scientific paper should reflect the state of the art. 

We oriented ourselves on the quickstart guide, however, if you have other suggestions, that would improve pybullets performance, feel free to mention anything. Currently the used settings are the following:

```python
self._p = bullet_client.BulletClient(connection_mode=pybullet.DIRECT)
self._p.setPhysicsEngineParameter(contactERP=0.2)
self._p.setPhysicsEngineParameter(globalCFM=2e-2)
```

and for objects: (pick and place scenario):
```python
for o in self.objects:            
            self._p.changeDynamics(o, -1, lateralFriction=0.45)
            self._p.changeDynamics(o, -1, spinningFriction=0.05)
            self._p.changeDynamics(o, -1, rollingFriction=0.001)
            self._p.changeDynamics(o, -1, restitution=0.0)
            self._p.changeDynamics(o, -1, linearDamping=0.04)
            self._p.changeDynamics(o, -1, angularDamping=0.04)
            self._p.changeDynamics(o, -1, frictionAnchor=1)
```

erwincoumans, 2020-10-08:
Heh, most of the 'scientific papers' are written by an author who has a physics engine in the race, so such 'paper' can better be described as an advertisement: the physics engine by the author always comes out best.
Bullet still has an OpenCL backend, we haven't enabled it at the moment, but we may do so in the future.

Tuning parameters depends on the scenario, there is not one 'best' set of parameters (otherwise we could get rid of them).

Simon-Steinmann, 2020-10-09:
> Heh, I'm a bit sceptical about benchmark papers, since most of the 'scientific papers about physics engine benchmarks' so far turn out to be advertisements: they are written by an author who has a physics engine in the race, and magically the physics engine by the author always comes out best

This became very apparent to us as well. We have no affiliation with any engine / simulation author. We are a consortium of several research institutes (DLR, Fraunhofer) and a few robotics and aerospace companies. The goal is to figure out, which engine and simulation is best suited for certain tasks. Thank you for your list of information. In a feature overview, some of these things will certainly be listed.

While it is not in the scope of this paper, I would be very interested in the implemented IK solution. Especially for RL, very fast IK is critical. So far I use IKFast, as it is several orders of magnitude faster than numerical solvers. Can you link me to some documentation? 
- Is it analytical or numeric? 
- Does it support any number of DOF? 
- Does pybullet include a motion planning framework akin to OMPL or MoveIt?

erwincoumans, 2020-10-12:
PyBullet's IK is numerical, using selectively damped least squares.

>> So far I use IKFast, as it is several orders of magnitude faster than
numerical solvers.

No, that is not my experience at all, IKFast was must slower than PyBullet.
I made a comparison between PyBullet and IKFast for the XArm 6.
In that case, IKFast often didn't produce a solution, and it was much
slower:
See
https://github.com/erwincoumans/xArm-Python-SDK/blob/master/example/wrapper/xarm6/xarm_sim.py

At the moment, no motion planning published, some projects using learning
for motion planning.
We also used Reflexxes 2, by Torsten Kroeger, for motion planning before,
also not released.
PyBullet does have various collision queries that can help motion planning.
We didn't expose the swept volumes in Python yet, should be easy to do.

We use PyBullet primarily for quadruped locomotion research, sim-to-real,
motion imitation etc, reinforcement learning etc,
and for arm grasping research and robot development. We also did some
projects for miniature car driving (MIT racecar etc).

Thanks!
Erwin

erwincoumans, 2020-10-12:
Timings were in the file: IK Fast was 300us versus PyBullet 150us, for the
IK on Xarm6.

erwincoumans, 2020-10-12:
(re-open to keep some discussion going)

BluePrintRandom, 2020-10-12:
There is a motion plan system for game agents called phase function neural
network, I believe making a plan with this and simulating it could be
supperior to IK and faster.

WilliamTambellini, 2020-11-03:
The question is very legit and early closing the ticket would be questionable.
Please find a recent paper confirming the need for GPU accelerated physics engines : 
"Local Optimization for Robust Signed Distance Field Collision"
https://mmacklin.com/sdfcontact.pdf
and a quick summary video:
https://www.youtube.com/watch?v=ooZ9rUYOFI4
First, has bullet3 an implementation of SDF (Signed Distance Field)?
Kind regards

closed 2021-04-26 by erwincoumans