bulletphysics/bullet3 #3106: Joint state is constant no matter the state of the joint.

MatthiasKohler, 2020-10-09:
I have a series of revolute joint, each joint additionally subject to a spring constraint. I can move the joints. However the joint state always is constant and does not change if the joint, for instance its angle, changes.  For one joint one of the state variables is nan.

[snake.zip](https://github.com/bulletphysics/bullet3/files/5354066/snake.zip)

pybullet build time: Oct  8 2020 00:10:04

erwincoumans, 2020-10-09:
useMaximalCoordinates=True doesn't support most features, such as getting the joint angles in getJointState.

erwincoumans, 2020-10-09:
I'll leave it open for now, in case we find time to implement getJointState for maximal coordinate rigid bodies.

closed 2021-04-26 by erwincoumans