bulletphysics/bullet3 #3118: Different textures (colors) on a cube?

trqminh, 2020-10-22:
Can I change the texture of a cube look like the image below by using pybullet.changeVisualShape and pybullet.loadTexture

[image]

trqminh, 2020-11-03:
I finally found out that I have to create a well-designed material as a .png file via using Blender or other 3D viewer tools.

closed 2020-11-03 by trqminh

louisdehaybe, 2022-03-29:
Hey, please could you show the file you used ? And also how you loaded it in bullet ? 
I have tried with some textures on internet but it basically replicates the whole png on each face of the cube instead of using each subset of the texture for each face..

Thanks !

trqminh, 2022-03-29:
@louisdehaybe 
Here is the [png file](https://drive.google.com/file/d/1rY-T7Nnu3rVUOyXSNXKWrDchuitGxuat/view?usp=sharing) I used to load the texture on my object. 
Actually, my colleague helps me with that, I think this is [one](https://www.youtube.com/watch?v=r5YNJghc81U) of her references

About the loading code, here is my code on creating an object instance in pybullet, actually when you have the right png file for the object, you can load the texture with its path and then apply changeVisualShape. Hope this helps:
```
import pybullet_data
import os

class StepStone:
    def __init__(self,
                 physics_client,
                 basePosition,
                 baseRPY,
                 useFixedBase=True,
                 globalScaling=1.0,
                 rgbaColor=None,
                 texture_path=None):
        self._p = physics_client
        self._p.setAdditionalSearchPath(pybullet_data.getDataPath())
        urdf = os.getcwd() + 'bureau/assets/step_stone/step_stone.urdf'
        self.id = self._p.loadURDF(urdf,
                                   basePosition=basePosition,
                                   baseOrientation=self._p.getQuaternionFromEuler(baseRPY),
                                   useMaximalCoordinates=0,
                                   useFixedBase=useFixedBase,
                                   flags=0,
                                   globalScaling=globalScaling,
                                   physicsClientId=0)
        if texture_path:
            texture = self._p.loadTexture(texture_path)
            self._p.changeVisualShape(self.id, -1, textureUniqueId=texture)
        if rgbaColor:
            self._p.changeVisualShape(self.id, -1, rgbaColor=rgbaColor)
```

louisdehaybe, 2022-03-29:
Thanks a lot for your quick answer. 
However, I get the same problem when I try with the texture you sent.  The whole picture appears on every face of the cube, instead of having each face covered with one colour. 

See below: 

[image]

Is this also what you get ? My goal is to have each face of the cube with one color. 

Cheers 
Louis

trqminh, 2022-03-30:
@louisdehaybe 
I think you misunderstood my answer.
As I mentioned, you need to first create your own png texture file that is suitable for your object,  The png file I provided is suitable for my object only so you cannot use that png to load for your cube (here is the [picture](https://drive.google.com/file/d/1OvuR2VPOfTByiye0RbYLDSCryLURIPZH/view?usp=sharing) of my object, that's why I have the corresponding png texture file). I also mentioned the youtube tutorial on how to create that kind of png file in my previous comment.

So in general, my suggestion is you load your urdf cube into the blender app and follow the youtube link that I mentioned to create a png texture file.

louisdehaybe, 2022-04-11:
Ok thanks I see now ! 
I checked your video, indeed he places a texture on the cube in blender but it does not show how to export the picture afterwards and I tried some things but it did not work :/

Akshay6077, 2024-03-27:
Hey any idea how could I import a terrain like material in Pybullet onto a simple platform like structure?